prusa3d/PrusaSlicer #2020: Facing errors on building Debug mode in VS2017 and VS2013

jayraaj, 2019-03-24:
### Version
Prusa 1.42.0 Alpha 7

### Operating system type + version
Windows 10

### Behavior
I am Facing errors on building Debug mode in VS2017 and VS2013 any buddy have built this in Visual Studio.

bubnikv, 2019-03-25:
Cool. What errors?

jayraaj, 2019-03-25:
I am facing linking errors and most of because wxwidgets.I have used the wxeidget 3.1.2 and for VS 2017 have built wxeidget on VS 2017

bubnikv, 2019-03-25:
Cool. A build log would be helpful as well.

jayraaj, 2019-03-25:
Actually Currently I have moved to Alpha 7 and building it on VS2017 Debug mode and facing the errors. I am able to build it in Release mode.

Also Attaching the Errors file.
[Alpha-7 Built on VS 2017 Debug Mode.txt](https://github.com/prusa3d/Slic3r/files/3005409/Alpha-7.Built.on.VS.2017.Debug.Mode.txt)

vojtechkral, 2019-03-26:
@jayraaj Are you following the [How to build](https://github.com/prusa3d/Slic3r/blob/aca2ffcdfbefecf00437d89a965d1ccdade567e6/doc/How%20to%20build%20-%20Windows.md) document or are you building on your own?
If you can build in Release mode but are having troubles linking dependencies in Debug mode, chances are your dependencies are built Release-only.

bubnikv, 2019-05-04:
no response, closing.

closed 2019-05-04 by bubnikv