prusa3d/PrusaSlicer #2021: [1.42.0-beta1] Infill % is overridden on startup

JohnOCFII, 2019-03-24:
### Version
_1.42.0-beta-1_

### Operating system type + version
_MacOS Mojave (10.14.3)_

### 3D printer brand / version + firmware version (if known)
_RailCore 300ZL - DuetWiFi v 2.02_

### Behavior
* _When loaded, this config's default infill % is overridden with a 20% infill_

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

Config attached.
[Railcore-config-export.ini.zip](https://github.com/prusa3d/Slic3r/files/3000926/Railcore-config-export.ini.zip)
[SquaredCircle 25mm-amf.zip](https://github.com/prusa3d/Slic3r/files/3000930/SquaredCircle.25mm-amf.zip)

FidelCapo, 2019-03-25:
I'm sorry, I don't understand your issue. Can you please describe it more detail with individual steps? Thank you

JohnOCFII, 2019-03-25:
Hello,

The profile has 10% infill as a saved setting.  When I launch Slic3r PE Beta1, and look at the printing profile settings, the infill setting is re-set to 20%.  I can change it back to 10%, and re-save the configuration, and quit Slic3r, but when I quit and re-launch Slic3r, the 10% setting has been replaced with 20%.  Two screen shots attached.

This configuration was exported from Slic3r-alpha 7 and imported into Slic3r-Beta0 (and now is read by Slic3r-Beta1).
[image]
[image]

FidelCapo, 2019-03-28:
Thank you for report. This issue should be fixed in next release.
Closing

closed 2019-03-28 by FidelCapo