prusa3d/PrusaSlicer #2022: Some problems with Lithophane still persist.

dzieva, 2019-03-25:
### Version
Slic3r 1.42.0-beta1+win64

### Operating system type + version
Windows 10 Pro

### 3D printer brand / version + firmware version (if known)
Marlin 2.0

### Behavior
* _Describe the problem_
Spaces without printing
[image]
* _Expected Results_
It should be like this one from S3D
[image]

_Is this a new feature request?_
no

#### Project File (.3MF) where problem]
[Lithophane_failed.zip](https://github.com/prusa3d/Slic3r/files/3003589/Lithophane_failed.zip)

supermerill, 2019-03-25:
thin walls activated?

dzieva, 2019-03-25:
Yes.
[image]

bubnikv, 2019-04-13:
When slicing with the prusa3d profiles, the gaps are not there. We have the "detect thin walls" feature disabled by default. 

Without the "detect thin walls" feature enabled, Slic3r allows a significant overlap between the last perimeter lines. With the "detect thin walls" feature, the gap fill shall kick in, filling the gaps with single lines. For some reason this is not happening on your print. I know @supermerill invested considerable time trying to improve the gap fill algorithm. Maybe he can use your model for his testing.

closed 2026-09-29 by prusaslicer-bot[bot]