prusa3d/PrusaSlicer #2023: [Feature Request] Increase size of Print Settings combobox on Print Settings Tab

Jebtrix, 2019-03-25:
### Version
v1.42.0-beta1

### Operating system type + version
Win7 PRO 64bit 

Request to increase the size of the combobox for Print Settings on the Print Settings Tab to match the size it has on the Plater Tab. The smaller combobox on Print Settings Tab always cuts off my print setting names.

[image]

[image]

Jebtrix, 2019-03-25:
Realized I'm actually asking to increase size for the settings combobox across all Tabs to match Plater since they share the same sizing.

bubnikv, 2019-04-13:
That makes sense. @YuSanka would you please take care of it?

bubnikv, 2019-05-21:
Implemented a long time ago with https://github.com/prusa3d/PrusaSlicer/commit/ae2c61160f15e5e8f49e98ff7c329a2ebc4a4df2. Closing.

closed 2019-05-21 by bubnikv