prusa3d/PrusaSlicer #2026: Can't select incompatible presets to change

MBadberg, 2019-03-25:
### Version
1.42.0beta+win64

### Operating system type + version
Windows 10 Pro
Version	10.0.17134 Build 17134

### 3D printer brand / version + firmware version (if known)
MK2S with latest Firmware

### Behavior
I figured out, that i have a wrong option in my user preset, so slic3r marks it as incompatible to my Printer preset. With any printer preset! I can't change to the incompatible preset. When i select it, slic3r selects automatically some compatible preset. 

_Is this a new feature request?_
I've found no similar issue

bubnikv, 2019-04-13:
Please zip and attach your config directory, and share with us, which combination of profiles cause you a headache.

rtyr, 2020-05-04:
No response (and old PrusaSlicer version). Closing.

closed 2020-05-04 by rtyr