prusa3d/PrusaSlicer #2029: [1.42.0-beta1] Vertical Scrollbar in Right Pane Coveres Widgits

Sembazuru, 2019-03-26:
### Version
Slic3r Prusa Edition
Version:   1.42.0-beta1+win64
Build:     Slic3rPE-1.42.0-beta1+win64-full-201903231838

### Operating system type + version
Operating System:    Windows
System Architecture: 64 bit
Windows Version:     Windows 10 (build 17763), 64-bit edition

### 3D printer brand / version + firmware version (if known)
N/A

### Behavior
If there is enough information in the right pane (my example has a large object list for a custom temperature tower) a vertical scroll bar needs to be displayed. The widgits in the right pane aren't re laid out so the far right widgits (the gear icons next to the preset pull-downs and the scroll bar of the object list) are obscured by the scroll bar.

This was also happening in the first beta, but since one could open the pulldowns by clicking anywhere on them, it wasn't as big of an issue. Now the new functionality is reduced. Also, my mouse's scroll wheel allows scrolling of the object list without needing access the scroll bar, but if a user doesn't have a scroll wheel, they can't scroll the object list.

See this screenshot:
[image]

Sembazuru, 2019-03-26:
I see this is being worked on. I just want to report in that I observed this on my HiDPI laptop, but on my normal DPI (LoDPI?) laptop the widgets did scale down properly to not be occluded.

YuSanka, 2019-03-27:
This Problem is concerned on initial size of an application frame not on HiDPI.

bubnikv, 2019-04-13:
It will be fixed in the upcoming beta2.

closed 2019-04-13 by bubnikv