prusa3d/PrusaSlicer #2030: Change/increase brim size for wipe tower.

Dutchy-79, 2019-03-26:
### Version
1.41.3

### Operating system type + version
Win10 64bit

### 3D printer brand / version + firmware version (if known)
MK3S MMU2S 3.7.0 1.05
### Behavior
Think I need a larger brim for the wipe tower

#### Project File (.3MF) where problem occurs
Several

Description:
I've started playing with the MMU2S and on two occasions the wipe tower started curling and came loose.
The bed (Textured sheet everyone wants) is clean, I cleaned with IPA and dish washing soap and problem stays the same.

Printing with PLA 205/210/215C 60/65/70C

I can rotated the purge block, Change width etc though not change the brim size in Slic3r PE.

bubnikv, 2019-04-13:
There was a bug in the brim for the wipe tower, where the brim was not fused to the wipe tower correctly. The current beta fixes that
https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-beta1

Does that solve your problem?

bubnikv, 2019-05-04:
bug fixed, no response. closing.

closed 2019-05-04 by bubnikv