prusa3d/PrusaSlicer #2031: Z-seam behavior bad in 1.42.0, good in 1.41.3

faboaic, 2019-03-26:
### Version
Slic3rPE-1.42.0-beta+win64-full-201903171406
Slic3rPE-1.42.0-beta1+win64-full-201903231838

### Operating system type + version
Windows 10
10.0.14393

### 3D printer brand / version + firmware version (if known)
Creality CR-10
Volcano clone, 0.6mm nozzle, PLA filament
Klipper

### Behavior
Same 3mf in 1.42.0-beta or beta1 leads to totally different behavior on Z-seam than Slic3rPE-1.41.3+win64-full-201902111723

I spent  half a spool of filament and it was a pain to find a (more or less...) good Z-seam with my test print. 
There was always much material missing on the seam.
At the end I had to add 1mm of restart distance for retract to keep the seam closed.

I could not believe it and opened the same 3mf in 1.41.3 and there was a huge Z-seam, like I expected with 1mm of restart distance. Far too much material.

* _Steps needed to reproduce the problem_

I think it has something to do with the 0.32mm layerheight and 0.6mm nozzle diameter.

Import attached 3mf (exported from 1.42.0-beta) to 1.42.0-beta or beta1 --> despite 1mm retract restart distance no mega-blob.

Import attached 3mf  to 1.41.3 --> mega-blob like expected.

* _Expected Results_
Identical behavior in 1.41.3 and 1.42.0-beta*

* _Actual Results_
Different behavior. 

Left: 1.42.0-beta (no big seam despite 1mm restart dist)
Right: 1.41.3 (blobby seam)

[image]

If you enter some "good in 1.41.3 settings" (e.g. restart distance = 0) you will have so much material missing with 1.42.0-beta*

_Is this a new feature request?_
no, it seems to be a bug.

If you look at the gcode and compare it between the versions there is one line at every layer change that makes a big difference:

[image]
Is wipe missing in 1.42.0-beta? 

#### Project File (.3MF) where problem occurs
I cannot upload .3mf, so I zipped the 3mf together with result gcode from 1.42.0-beta and from 1.41.3

[z-seam-3mf-and-gcodes.zip](https://github.com/prusa3d/Slic3r/files/3008862/z-seam-3mf-and-gcodes.zip)

........
And yes... it really took me so much tries to find a Z-seam that was not missing material or other issues ;-)
[image]

faboaic, 2019-03-26:
retested with @supermerill 's fork, there I see the expected seam with extra length on restart = 1mm.
So his version is not affected by the problem.
https://github.com/supermerill/Slic3r/releases/tag/1.42.0-alpha7.0

closed 2019-03-31 by faboaic

supermerill, 2019-03-31:
Something changed between alpha7 and beta1 in seams or not?

faboaic, 2019-03-31:
Not really sure...
I am doing extensive testing at the moment with two printers and three slic3rs to get to a final statement.

At the first tests there have been too many unknowns in the variable:
First use of volcano clone with 0.6mm nozzle.
Test print was single wall, smaller than nozzle diameter and Extrusion Width.
Not perfectly calibrated pressure advance in Klipper.
Changed from known ABS filament to cheap crap PLA which I have no experience with.
...

Since the ticket I changed back to e3d clone with known pressure advance settings, old and known as good settings in slic3r, etc.

Sorry for all the confusion... That's why I closed my ticket in the hope that I was doing something wrong in my tests...

I will check things up and create a new ticket with less confusion.

It seems that I now have some settings where I get good results with slic3r prusa edition 1.42.0b1 and 1.40.1.
Same settings with your 1.42.0a7 still issues...

Maybe there is an additional issue with 3mf import...

bubnikv, 2019-04-01:
From the slic3r PE 1.42.0-beta release log:

   - Fixed one more regression issue worsening handling of broken models
   compared to the Slicer PE 1.41.3 #1869
   <https://github.com/prusa3d/Slic3r/issues/1869> #1920
   <https://github.com/prusa3d/Slic3r/issues/1920> #1960
   <https://github.com/prusa3d/Slic3r/issues/1960>.

This may have had some effect on the seams.

We will be thankful, if you could publish your data on Slic3r PE with the
Volcano and high throughput nozzles.

faboaic, 2019-04-02:
Sorry for being off topic... That's why I closed the ticket before.
Main reason for bad results was bad calibrated pressure advance in Klipper with totally wrong retract settings.
Still not perfect but now a status I can build upon.

First result with e3d clone, 0.4mm nozzle, cheap PLA, Klipper, creality cr10.
All used the same 3mf as import.
Results partially equal, partially totally different.

Interesting while comparing the gcode files e.g. with notepad++ with compare plugin...

[image]
[image]
[image]
[image]
[image]

[githubfiles#2031.zip](https://github.com/prusa3d/Slic3r/files/3032750/githubfiles.2031.zip)

The results with the thin walled tubes are bad in every Slicer that I tested until now... So I took my cut heavy duty cr10 fan wing retract test which I am familiar with.

guestisp, 2019-04-04:
I don't think that a wrong Klipper pressure advance is the culpit. I'm using Marlin2, without pressure advance, and I'm experiencing the same issue on the seam. My printer (Geeetech A20 with E3D v6, bondtech clone, capricorn tube clone) is calibrated (thin-wall perfect, steps perfect and so on) but with beta1 I have some holes in the seam that i've never experienced before.

bubnikv, 2019-04-05:
If it is a single wall print, then it is printed by Slic3r using the thin
wall detection algorithm.

The seam hiding and nozzle wipe algorithms were tested and fine tuned on
normal prints, not on these single wall prints. I suppose Slic3r could
handle these prints better. Does it help, if you disable wipe?

guestisp, 2019-04-05:
It's a 3 perimeters print.
I didn't use the wipe in that print

guestisp, 2019-04-05:
Please ignore the horizontal line, i've broke the part trying to remove It from the bed.
Here you can see the seam. It's huge

[image]

supermerill, 2019-04-05:
> 
> 
> It's a 3 perimeters print.
> I didn't use the wipe in that print

I downloaded your model & settings. when using them, It's a single thin wall, no perimeters. Maybe i did something wrong?

guestisp, 2019-04-05:
> I downloaded your model & settings: when using them I have a single thin wall, no perimeters. Maybe I did something wrong?

Downloaded from where? I've not posted the print setting for that model.
All my prints are with 3 perimeters, i've never printed with less (except the thin wall calibrations)

supermerill, 2019-04-05:
z-seam-3mf-and-gcodes.zip

Can't test or say anything about the other one.

guestisp, 2019-04-05:
> z-seam-3mf-and-gcodes.zip

this is not mine.

The model in the photo is this: https://www.thingiverse.com/thing:1781090
printed with 3 perimeters, seam aligned, extrusion width 0.40 (calibrated), nozzle 0.40, infill 20% gyroid, slic3r beta1

guestisp, 2019-04-21:
No more updates on this?