prusa3d/PrusaSlicer #2032: Thin thick bridges

espr14, 2019-03-26:
Issues with bridges:
1. slackness
2. gaps between lines
3. non-homogeneous (rough) surface

[https://youtu.be/BeyMhXaVkbc](https://youtu.be/BeyMhXaVkbc)
This video shows that slightly slower speed and lower extrusion is a key to a good bridge. However, is the lower extrusion compensated by Slic3r PE by more lines?

The thinner the line, the easier it carries itself and it cools faster. Also gaps between lines won't be so noticeable. What about changing bridge code generation? Ideal printing case:
1. Print thinner lines (underextrusion).
2. Print lines more close to each other (compensates underextrusion and avoids gaps).
3. Print in smaller layer height (next layer can fill gaps of the previous one).

bubnikv, 2019-04-13:
Bridging flow 0.1 (aka 10% of the nozzle diameter):
[image]

the bridging flow is certainly respected in the spacing.

The bridge threads are separated by a hard coded value:
// Extra spacing of bridge threads, in mm.
#define BRIDGE_EXTRA_SPACING 0.05

espr14, 2019-04-13:
Nice. So 2 of 3 are solved already. Now:

- What about automatic decreasing layer height for bridges?
- What about bringing the hard coded BRIDGE_EXTRA_SPACING into the expert setting of Slic3r?

bubnikv, 2019-09-05:
> What about automatic decreasing layer height for bridges?

I don't understand that request.

espr14, 2019-09-05:
Let's the layer height is 0.2. Bridge can be printed in 0.1 so the bridge is printed twice per standard layer. Something like different layer height for sparse infill.

bubnikv, 2019-09-05:
> Bridge can be printed in 0.1 

Bridge is not printable at 0.1 (is it diameter or height)?

I cannot make sense of what you are trying to explain. Maybe an image would help?

espr14, 2019-09-05:
[image]
 Similar for bridges but instead of multiplying layer height it would divide the layer height.

So in case the coefficient is "2", the object is printed say 0.2 mm layer height. Bridges are printed with 0.2/2 = 0.1 mm layer height but twice which results in 0.2 height. First bridge layer(s) first, then the rest of the object, then the last layer of bridges.

In case of coeff. "3", and lay.height 0.2, bridges will be printed in 0.2/3 = 0.6666 mm layer height. 

4, 0.2 =>0.05 ......

bubnikv, 2019-09-06:
Would you please explain what problem are you trying to solve? I believe you are Czech, you may write in Czech language if it is easier for you.

espr14, 2019-09-10:
Problem: supported and unsupported bridges look bad. 

Here are unsupported bridges. Bridge flow ratio was kept the same (1). First has 0.1 mm layer height, second 0.2 and the third has 0.2 mm with 0.55 gaps in model. The perpendicular shift is caused by contraction of the 2nd bridge layer.

First two look bad. The last one looks better but it is much weaker. One possibility is to print odd and even lines separately. The 2nd layer could be in the same direction with higher (full) flow so it would bring strength without perpendicular shift.

[image]

closed 2026-09-29 by prusaslicer-bot[bot]