prusa3d/PrusaSlicer #2033: Test patterns

espr14, 2019-03-26:
Slicer should include some test patterns which will help with slicer settings. I know that Slic3r PE has lot of presets. But it is used on different printers with different materials which are not in the list. There are lot of test models around but, typically, they are not intended for tuning of parameters. They just tell you that your setting is OK or not. It does not tell what each setting should be set to.

Let's come with specific test for nearly each setting. Each test will print few (3, 5, ...) samples with different setting of the specific item. For example it will print bridges each with different print speed. Some of them with lower speed, some of them with higher speed and one with the currently set speed. There will be a number or mark on each bridge to distinguish results after removing from bed. After print user looks at the results and sets the appropriate value to the bridge speed.

Printing all the tests will take time but sometimes you want to tune your printer with speific filament so every result will look great.

faboaic, 2019-03-27:
Great idea.
Something like temp tower, retract tower, etc. would be perfect.

It is possible to create a temptower manually ( gcode postprocessing , like {if layer_num ==99934} M104 S250 {endif} )  ... but that is not so easy for beginners.

A "change at z" like in cura would be a great option in combination with test patterns...

You could print two tubes next to each other and change retract from 0 to 10mm to find your optimal settings.

SphaeroX, 2023-12-28:
I would like to revisit this idea but transform it. Namely, I was just about to print a temp tower and thought I'd check the shapes gallery, but there's only a Benchy, bunny, etc. This made me wonder why the typical calibration files aren't included. It would make sense to add a temp tower and so on from the beginning. Sure, I can add it myself, but what do you think of the idea of doing this from the start?

[image]

closed 2026-09-29 by prusaslicer-bot[bot]