prusa3d/PrusaSlicer #2034: Automatic variable layer height for lower roughness

espr14, 2019-03-26:
Big layer height produces uniform surface but it cannot handle overhangs over 45° well. Small layer height will handle over 60° but the surface is rough. It would be great to set some matrix which defines which layer height should be used for which angle. This would preserve nice look on easy parts while it will handle extreme overhangs without support (which causes even worse surface).

Cura has automatic layer height but it just counts some trigonometry to avoid noticeable steps know from big layer height. It does not deal with roughness because it is more about printer and filament which cannot be computed.

bubnikv, 2019-05-04:
Duplicate of https://github.com/prusa3d/Slic3r/issues/466 ?

espr14, 2019-05-04:
Kind of. But this is more about quality of thin layers. The thinner layer the lower surface quality.

SvetoBobr, 2019-10-20:
#466 request is more about a dialog. 

Current smooth layer height is nice but useless in command line. So the feature as it implemented in the original slic3r is really important.

bubnikv, 2021-09-23:
Automatic variable layer height was implemented quite some time ago.
If you think it is insufficient, please provide a pull request.

closed 2021-09-23 by bubnikv