prusa3d/PrusaSlicer #2035: Option to remove machine limits from g-code

guestisp, 2019-03-27:
How can I remove the whole "Machine limits" values from g-code ?
I'm using the default from the firmware and I would like to get rid of a bouch of lines in the gcode that could lead to misbehaviours. In example, updating a firmware and printing an old gcode with M201..M205 set or viceversa changing one value as mistake 

Additionally, with Marlin v2, Jerk was removed so, M205 is not needed anymore.

I've tried to set all values to 0 but in tal case, everything is still exported, as zero.
How can I remove all of these ?

Timocop, 2019-04-01:
Machine limits should be disabled by default if people choose 'custom printer' in the configuration. Since the default values only apply to Prusa Orginal printers which may lead to skipping/grinding on other printers with those high values. Even tho its pretty easy to put in your values in the machine limits section beginners may get confused. Because the machine limits section is not visible in Simple-Mode and miss configuration can still happen.

The first time i tried Slic3r PE and printed with default machine limits, my Z-Axis steppers skipped way too much.

As for print time calculation i dont think using Prusa Orginal defaults is as bad.

guestisp, 2019-04-01:
Or, if set to 0, the machine limits should be removed from the gcode and let the firmware handle them by itself. You can't have a printing speed as 0, so setting 0 values could do the trick.

bubnikv, 2019-04-13:
Duplicate of https://github.com/prusa3d/Slic3r/issues/1212

closed 2019-04-13 by bubnikv

gretel, 2019-06-13:
> Or, if set to 0, the machine limits should be removed from the gcode and let the firmware handle them by itself. You can't have a printing speed as 0, so setting 0 values could do the trick.

@guestisp setting `0` only "helps" for the speed limits, all others are not omitted but set exactly as defined:

```
M201 X0 Y0 Z0 E0 ; sets maximum accelerations, mm/sec^2
M203 X0 Y0 Z0 E0 ; sets maximum feedrates, mm/sec
M204 P0 R0 T0 ; sets acceleration (P, T) and retract acceleration (R), mm/sec^2
M205 X0.00 Y0.00 Z0.00 E0.00 ; sets the jerk limits, mm/sec
M205 S0 T0 ; sets the minimum extruding and travel feed rate, mm/sec
```

```