prusa3d/PrusaSlicer #2037: no public CI

jhoblitt, 2019-03-27:
I personally dislike contributing to a project that doesn't have visible CI in place for PRs.  As a bonus, an in-repo CI config also serves as a provably working set of build instructions.

bubnikv, 2019-03-28:
> I personally dislike contributing to a project that doesn't have visible
CI in place for PRs

Please contribute a CI, that would produce builds for all three major
platforms for PRs. Thank you.

kkio, 2019-07-02:
Would there be any interest in using [Azure Pipelines](https://azure.microsoft.com/en-us/services/devops/pipelines/)?

It was the only free for open source service I could find that supported all three major platforms.

Only caveat would be that the Mac build image doesn't support the older 10.9 SDK used to build the MacOS binaries.

Are there any further wants/requirements besides "have CI job run on PR"? I'm still doing some initial investigation but I might want to take a stab at this.

vojtechkral, 2019-07-02:
> Only caveat would be that the Mac build image doesn't support the older 10.9 SDK used to build the MacOS binaries for release.

That could probably be fixed one way or another. Using the older SDK is just a matter of downloading an archive, although I have no idea about the legality of such a thing (IANAL). It might also be possible for us to relax the 10.9 requirement, depending on how much dependencies allow...

kkio, 2019-07-02:
For anyone curious, you can find the SDKs included [here](https://github.com/Microsoft/azure-pipelines-image-generation/blob/master/images/macos/macos-10.13-Readme.md#installed-sdks).

There are tools like [XCodeLegacy](https://github.com/devernay/xcodelegacy) that we can try using to script getting the SDK.

userosos, 2019-07-23:
Update status please. It work now?

spuder, 2019-12-28:
Github Actions now supports all 3 platforms. It will give visibility into the build process and is free. 

https://help.github.com/en/actions/automating-your-workflow-with-github-actions/software-installed-on-github-hosted-runners#xcode

It still does not solve the problem of using an older SDK. Maybe it's time to update.

supermerill, 2020-01-09:
I make three github action for building for ubuntu, windows & mac.
https://github.com/supermerill/Slic3r/tree/master/.github/workflows

but i don't know how to properly package the mac one.

spuder, 2021-12-09:
Related conversation: https://github.com/prusa3d/PrusaSlicer/issues/7412

centerorbit, 2021-12-09:
I'd be interested in starting a GitHub Actions pipeline for this project. Being it's hosted on Github, I think Actions makes the most sense.

camdenorrb, 2022-06-22:
CircleCI would also be a good option

closed 2026-09-29 by prusaslicer-bot[bot]