prusa3d/PrusaSlicer #2038: Bridge infill

espr14, 2019-03-27:
You can find bridge infill in preview window. If you go to Print settings -> Speed there is no bridge infill. There are settings for infills and bridge. Which setting does it use? Similarly, you can set bridge settings but there is no bridge in the preview window. Could you unify settings and preview?

[image]

bubnikv, 2019-04-13:
> You can find bridge infill in preview window. If you go to Print settings -> Speed there is no bridge infill. There are settings for infills and bridge. Which setting does it use?

bridging infill uses bridging parameters.

> Similarly, you can set bridge settings but there is no bridge in the preview window. Could you unify settings and preview?

I don't understand.

espr14, 2019-04-13:
Let's take it from another side. There are 4 types:
1. Bridge perimeter
2. **Bridge infill**
3. Overhang perimeter
4. **Overhang infill**

Only those in bold font are in the preview (picture above). On the other hand, in the Print settings (ctrl-2 tab) there is only settings for bridge (first 2 items). No settings for overhangs. But I suppose the overhangs use the same settings as bridges. That's confusing.

Bridges (infill and perimeters) are easier to print then overhangs. It would be nice to distinguish these in the color. On the other hand, it does not matter so much if it is bridge perimeter or infill so you don't need to distinguish these in colors (to keep number of colors low).

closed 2026-09-29 by prusaslicer-bot[bot]