prusa3d/PrusaSlicer #2039: Broken external perimeter

ochm, 2019-03-27:
### Version
1.42.0-beta+Win64-full-201903171406

### Operating system type + version
W8.1

### 3D printer brand / version + firmware version (if known)
Prusa i3 MK3

### Behavior
External perimeter is broken on some places, but in other slicers (Simplify3d, repetier-host) is object OK

_Is this a new feature request?_
No

#### Project File (.3MF) where problem occurs
[Box_Intersecting_Rods.zip](https://github.com/prusa3d/Slic3r/files/3015488/Box_Intersecting_Rods.zip)
[image]

ochm, 2019-03-28:
In version 1.41.2 are perimeters OK.

glowfishDE, 2019-03-29:
It looks like i have the same issue with external perimeters that break:

https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f61/slic3r-prusa-edition-1-42-0-beta1-t28595-s20.html

bubnikv, 2019-04-01:
The holes you are seeing in the preview are real, and they are there to
hide seams. They were always there in Slic3r, they just were not displayed
before you generated the G-code. Without these tiny gaps you get bulges
around the seams.

ochm, 2019-04-01:
I can't agree, the holes are in print too, IMHO should not be in print.

dzieva, 2019-04-01:
> The holes you are seeing in the preview are real, and they are there to hide seams. They were always there in Slic3r, they just were not displayed before you generated the G-code. Without these tiny gaps you get bulges around the seams.
> […](#)

Do you believe that this is also a tiny gaps?

[image]

bubnikv, 2019-04-01:
@dzieva Please do not mix apples with oranges. Your issue is certainly different from the original one. If you believe there is an issue, please open a new issue and provide all possible information, mainly the project file (3MF file). Thank you.

bubnikv, 2019-04-01:
@ochm We are printing with the beta a lot and we do not see any difference compared to the currently released Slic3r 1.41.3 PE.

@jindrichbenes would you please chime in? Thank you.

dzieva, 2019-04-01:
> @dzieva Please do not mix apples with oranges. Your issue is certainly different from the original one. If you believe there is an issue, please open a new issue and provide all possible information, mainly the project file (3MF file). Thank you.

I hoped we'd have the 1st of April.

Sorry, please.

jindrichbenes, 2019-04-03:
We were unable to observe a difference between Slic3r 1.41.3 and 1.42.0-beta1.

[image]
[image]

bubnikv, 2019-04-04:
Thanks @jindrichbenes for investigation. Could we close this issue then?

dzieva, 2019-04-12:
> > The holes you are seeing in the preview are real, and they are there to hide seams. They were always there in Slic3r, they just were not displayed before you generated the G-code. Without these tiny gaps you get bulges around the seams.
> > […](#)
> 
> Do you believe that this is also a tiny gaps?
> 
> [image]

Incredible things happen by disabling "Detect thin walls", it looks like it works the other way around.
The gap is gone.

bubnikv, 2019-04-13:
@ochm described seams, which are a property of the FDM technology. Closing.

closed 2019-04-13 by bubnikv