prusa3d/PrusaSlicer #2040: How to improve travel movement?

dzieva, 2019-03-28:
### Version
Slic3r 1.42.0-beta1 - Prusa Edition

### Operating system type + version
Windows 10 Pro 64

### 3D printer brand / version + firmware version (if known)
Marlin

### Behavior
* _Describe the problem_
Travel not realistic
[image]

* _Expected Results_
It could do the same as Simplify3D and Cura.
[image]

Timocop, 2019-04-01:
Enable "Avoid crossing perimeters" in "Print Settings > Layers and perimeters" in Expert-Mode. This should keep the nozzle within the infill area. Just like "Combing: Within infill" in Cura.

dzieva, 2019-04-02:
I had already tried using "Avoid crossing perimeters", but a nonsense of slow, often does not even finish slicing.
I also hope that someday they will improve it.

supermerill, 2019-04-03:
duplicate of https://github.com/prusa3d/Slic3r/issues/1934 ?

Timocop, 2019-04-03:
I found a solution for your slow slicing with "Avoid crossing perimeter" enabled. The problem are high polygone models. The resolution is set 0 by default (unlimited) which makes slicing complex models alot harder and slower. Since it loops though all perimeters and finds the perfect center. You should set the resolution to 0.01 (Cura default) this shouldnt sacrifice any detail but redues slicing time by ~50% in my compare tests.

bubnikv, 2019-04-04:
Thanks for the hint on reducing the resolution of slices. We have had some
bad experience with this feature, introducing visible artifacts by
simplifying successive layers independently from each other, thus reducing
a point on one layer while keeping a point of the same triangle edge on the
other layer. We may re-evaluate the feature for small reduction rates
though.

Timocop, 2019-04-04:
Do you mean variable layer high? I dont realy understand, sorry.
I only get visible artifact (holes, ripped walls) when having the flawed "Detect thin walls" detection enabled. Havnt seen any issues using Resolution 0.01. Do you have an example?

dzieva, 2019-04-05:
Timocop, thanks.

I did what you said. and it was possible to slice "after a long time", but the problem continues, nothing has changed.

[image]

Timocop, 2019-04-05:
Do you mind sharing the model? I dont think there will be a solution since 'Avoid crossing perimeters' behaves like Curas "Combing: Within infill" and your models doenst seem to have any infill area.
If you realy want combing that bady, you have to create more infill space (Less skin, more gap fill).

supermerill, 2019-04-05:
Gap fill is not infill. Don't put your expectation too high.

Timocop, 2019-04-05:
In the orginal slic3r gap fill is under the infill section. So i thought its infill. Guess not. Which doesnt make any sense to me.

supermerill, 2019-04-05:
It's done with the perimeters, internally, stored in the same collection.

dzieva, 2019-04-05:
> Do you mind sharing the model/config? I dont think there will be a solution since 'Avoid crossing perimeters' behaves like Curas "Combing: Within infill" and your model doenst seem to have much infill area.
> If you realy want combing that bady, you have to create more infill space (Less skin, more gap fill).

Here it is, it takes a lot of patience.

[slice3rPE_model&config.zip](https://github.com/prusa3d/Slic3r/files/3049345/slice3rPE_model.config.zip)

bubnikv, 2019-08-25:
The initial phase of the "avoid crossing perimeters" feature has been sped up with 3addae3, the feature is now also significantly faster.

This particular model took 60s to generate G-code on my computer with the "avoid crossing perimeters" feature on, taking 5h 20m to print, while it took 20s to generate G-code with the "avoid crossing perimeters" feature disabled, taking 5h 16m to print.

Closing this one.

There is still https://github.com/prusa3d/PrusaSlicer/issues/2213 for a reference of a complex object, which took 7 minutes to export the G-code with the "avoid crossing perimeters" function enabled.

closed 2019-08-25 by bubnikv