prusa3d/PrusaSlicer #2041: slic3r prusa

St-Clair, 2019-03-28:
### Version
_Version of Slic3r Prusa Edition used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

bubnikv, 2019-04-13:
Does it happen with the 1.42.0-beta? I was not able to reproduce the issue.
https://github.com/prusa3d/Slic3r/releases/tag/version_1.42.0-beta1

St-Clair, 2019-04-13:
I have reinstalled my OS since originally posting this issue, I was able to still replicate it in the current live build of 1.41.3 on a fresh install, downloading and testing in 1.42.0-beta1 does not replicate.

Appears fixed in beta update, closing issue.

closed 2019-04-13 by St-Clair