prusa3d/PrusaSlicer #2043: 1.42.0 alpha7 pop up (cutting etc) dialog boxes are way too small

Mirarkitty, 2019-03-28:
### Version
1.42.0 alpha 7

### Operating system type + version
Linux, "retina" 4k screen on 15"

### Behavior
* dialog box font size is not following system settings and is impossible to read 

[image]

ctilley83, 2019-03-28:
Have you tried this in beta 1?  You’re two versions behind.

Mirarkitty, 2019-03-28:
I did now, nothing changed, issue remained.

bubnikv, 2019-03-28:
Did you compile Slic3r yourself? What version of wxWidgets and GTK?

Mirarkitty, 2019-03-29:
No, this was the .AppImage. I still can't compile due to missing libraries in my dist. I'm not sure why those dialog boxes are not the normal GTK that seems to work fine?

vojtechkral, 2019-03-29:
> I'm not sure why those dialog boxes are not the normal GTK that seems to work fine?

That's because they need to be rendered in the 3D scene. As far as I know you can't do that with either wxWidgets or GTK.

In any case, the font size is probably not set correctly for the imgui dialogs, I'll have a look ...

vojtechkral, 2019-03-29:
@Mirarkitty this should be fixed in the next release...

closed 2019-03-29 by vojtechkral

vojtechkral, 2019-03-29:
@Mirarkitty would you please re-test with the following build?

[Slic3rPE-1.42.0-beta1+85-linux64-full-g3cfb234-201903291507.tar.gz](https://github.com/prusa3d/Slic3r/files/3023331/Slic3rPE-1.42.0-beta1%2B85-linux64-full-g3cfb234-201903291507.tar.gz)

Mirarkitty, 2019-03-30:
It's correct now, thanks.
[image]

I was just wondering why it wasn't GTK because it's a bit UX confusing to mix widgets.

vojtechkral, 2019-04-01:
Ok, thanks for testing!

commit 3cfb234e (2019-03-29): imgui: Get font size from wxWidgets, fix #2043