prusa3d/PrusaSlicer #2045: 1.42.0-beta1: send to printer vanished

Mirarkitty, 2019-03-29:
### Version
1.42.0-beta1 

### Operating system type + version
Linux AppImage

### 3D printer brand / version + firmware version (if known)
MK3 MMU2 with OctoPrint

### Behavior
* In Alpha7, "slice now", "export gcode" and "send to printer" was always visible
* In Beta1, "slice now" needs to be pushed before "export gcode" shows up
* "send to printer" never shows up
-- regression?

beta1:
[image]
[image]
[image]

In alpha7:
[image]

I have tried resetting the settings and filling in details again.

Mirarkitty, 2019-03-29:
Also tried cloning .Slic3rPE-alpha to .Slic3rPE-beta.

YuSanka, 2019-03-29:
It's funny, because of a "Send G-code" button is a same "Send to printer" button, but renamed ;) Just try to push it :)

JohnOCFII, 2019-03-29:
That change in behavior was explained in the release notes for the first Beta release:

> The two "Slice now" & "Export" buttons are now merged into a single button, changing its text and purpose based on the slicing status.

Mirarkitty, 2019-03-30:
I see. But why isn't it there before slicing?

cmburg, 2019-03-30:
because the file has to be sliced before it can be sent. Have you disabled the background processing in the preferences?

Mirarkitty, 2019-03-31:
Nope. Literally same setting for beta1 and alpha7, in alpha7 "send to printer" showed up at once.

bubnikv, 2019-04-01:
Does the current behavior cause you any troubles? Does it break your
workflow?

Mirarkitty, 2019-04-06:
Yes, I usually just hit "send". Not slice first. Most of my workflow is rapid prototyping (openscad->stl->reload->send), any step less is an improvement.

Mirarkitty, 2019-05-13:
Closing since this seems to work fine now with "send gcode" although "send to printer" was clearer.

closed 2019-05-13 by Mirarkitty