prusa3d/PrusaSlicer #2047: Mesh Calibration "Some problem encountered Z levelling enforced".

RosieTheRabbit, 2019-03-29:
### Version
_Version of Slic3r Prusa Edition used goes here_

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

Slic3rPE - 1.42.0-beta+win64-full-201903171406

### Operating system type + version
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known)
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

Windows 10 Home Edition (64 bit)

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
  * _If this is a command-line slicing issue, include the options used_
* _Expected Results_
* _Actual Results_
  * _Screenshots from __*Slic3r*__ preview are preferred_

_Is this a new feature request?_

Z Axis Mesh Levelling at start of print results in "Some problem encountered Z levelling enforced".

The printer goes through the calibration wizard with the result of "Congratulations the X and Y axes are perpendicular" or similar.  There are no corrections to make and the printer is accurately set up.  This has been checked by measuring and by the printer itself.

Once the printer has heated up, the print head goes to home then begins the mesh calibration process.  Assuming that 7x7 has been selected, the printer will get through approximately 45 or so calibration points then will stop with the error message "Some problem encountered Z levelling enforced".  If the setting is 3x3 then the printer will get through 7 or 8 points before stopping with the same error message.

The only option at that point is to go through the process dictated by the firmware - run the Z axis to the top of travel then let the printer go through mesh levelling again.  However when the printer gets to near the end of the process the same stop and error message occurs.

If I set the number of times the Pinda scans each point to 1, the printer will get through the mesh levelling process 9 times out of 10.  Setting the number of scans to 2 or more results in the stop and error message.

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

Problem occurs after printer pre-heat at start of print cycle when mesh levelling is in progress.  The printer will stop at the 45th or 8th point and display the same error message.

bubnikv, 2019-04-01:
Please report this issue to the firmware team
https://github.com/prusa3d/prusa-firmware/issues

or to the Prusa3D support. Thank you.

closed 2019-04-01 by bubnikv

RosieTheRabbit, 2019-04-01:
My apologies. Will send to firmware team. 

Thanks

Jim.