prusa3d/PrusaSlicer #2048: Can't slice a thin wall test

guestisp, 2019-03-29:
beta1

The attached STL can't be sliced properly. no error, just an empty gcode. Even the layer preview doesn't show anything.

[0.4mm-thin-wall.zip](https://github.com/prusa3d/Slic3r/files/3022970/0.4mm-thin-wall.zip)

BerndJM, 2019-03-29:
"Detect thin walls" enabled?

guestisp, 2019-03-29:
It's disabled. I have to calibrate the printer, I think that enabling that would mess with the results. I just need a single line as perimeter.

BerndJM, 2019-03-29:
Assuming your wall width is exactly 0.4mm this has to be detected as a thin wall.
Read the help text about this option and also read the recommended wall widths in the section "vertical shells".
And why not simply give it a try. Print one with "detect thin walls" and look at the result ...

guestisp, 2019-03-29:
I've not tried, because i'm able to print a different calibration cube, that is still 1 wall. Only this STL can't be printed.

My biggest concern is that `thin_wall` option would add some artefact on a calibration cube, maybe by adding lines or altering the wall width, that's why i've disabled it

bubnikv, 2019-04-01:
This cube cannot be printed without a "detect thin walls" option, it works with the "detect thin walls" option.
Closing.

closed 2019-04-01 by bubnikv

guestisp, 2019-04-01:
"detect thin walls" doesn't add artefacts ? Can I safely use that with a single line wall that has to be exactly as the extrusion width ?