prusa3d/PrusaSlicer #2051: "Original Prusa i3 MK# MMU2 Single" generates a "T0" after start G-Code

tellblom, 2019-03-31:
### Version
1.42.0-Beta+win64-full

### Operating system type + version
Windows 10 (1809 Build 177363.379)

### 3D printer brand / version + firmware version (if known)
Prusa MK3S with MMU2S

### Behavior
When using profile "Original Prusa i3 MK# MMU2 Single" it generates a "T0" after "Start G-Code"
[image]
 and this causes the printer to choose filament 1 whatever filament you choose to print with in the dialog.

In Slic3r PE 1.41.3 this "T0" is absent and it works as it should

#### Project File 
[Harley single color__.3mf.zip](https://github.com/prusa3d/Slic3r/files/3026462/Harley.single.color__.3mf.zip)

supermerill, 2019-04-01:
see also https://github.com/prusa3d/Slic3r/issues/1796

lukasmatena, 2019-08-21:
I could not reproduce it and I'm now quite certain this is the same as https://github.com/prusa3d/PrusaSlicer/issues/2166. The T0 command would be added only if printer profiles were recently switched from MMU to MMU Single.

I'm closing the issue. If anyone disagrees, I can reopen it.

closed 2019-08-21 by lukasmatena