prusa3d/PrusaSlicer #2052: Placeholder for extrusion width 0 if using default width

faboaic, 2019-03-31:
### Version
Slic3rPE-1.42.0-beta+win64-full-201903171406
Slic3rPE-1.42.0-beta1+win64-full-201903231838
all versions

### Operating system type + version
Windows 10
10.0.14393

### Behavior

I use default extrusion width for my extrusion and depending on the part I use percentage of this value or absolute value.

[image]

I expect to be able to use the default extrusion width as placeholder for including it in the gcode filename.

e.g.:
`[timestamp]_[input_filename_base]_[layer_height]h_[extrusion_width]w_[retract_length]retr_[nozzle_diameter]n_[temperature_0]c-[bed_temperature]c_[perimeters]p_[top_solid_layers]tb_[fill_pattern]_pla-cr10-s3r.gcode`

Is:
`20190331-133751_stlfilename_0.36h_0w_1.2retr_0.6n_200c-54c_3p_4tb_grid_pla-cr10-s3r.gcode`
Expected:
`20190331-133751_stlfilename_0.36h_0.66w_1.2retr_0.6n_200c-54c_3p_4tb_grid_pla-cr10-s3r.gcode`

_Is this a new feature request?_
I think it is more kind of a bug that wrong extrusion width (not default width) is used for the placeholder "extrusion width".
Otherwise it could be a feature request for a placeholder like "default_extrusion_width".

bubnikv, 2019-04-13:
This is very much the same issue as https://github.com/prusa3d/Slic3r/issues/930 and https://github.com/prusa3d/Slic3r/issues/1144

bubnikv, 2020-02-04:
Implemented with 7d7f66aca6e445ae235252af5f544ef7144e19e1

Please note that the old placeholder syntax [variable_name] still produces the old, unsubstituted result. If you put 50% there, you will get 50%.
On the other hand, the new placeholder syntax {variable_name} will produce numbers as close as possible to what the slicing back-end works with. There may be some discrepancies of extrusion width dependent on layer height if variable layer heigth is in action though.

Closing. It will be part of the upcoming PrusaSlicer 2.2.0-alpha4.

closed 2020-02-04 by bubnikv