prusa3d/PrusaSlicer #2054: SL1 suggenstions for Slic3r PE

port513, 2019-04-01:
### Version
1.42 Beta1

### Operating system type + version
Booth Windows and Linux

### 3D printer brand / version + firmware version (if known)
SL1

### Behavior
Make the model hollow with drainage holes or make the model hollow with open bottom.
And a few other things important to SLA-printing, don't want to use more than one software if not needed so to speak.

_Is this a new feature request?_
Yes

bubnikv, 2019-04-01:
SLA hollowing with drainage holes is in the planning, however not for this release.

> And a few other things important to SLA-printing, don't want to use more than one software if not needed so to speak.

We will also implement the few other things important to SLA-printing once you share these with us.

port513, 2019-04-02:
This is my wishlist

1. Hollow model with or without draining holes solid or open bottom on the model etc.
2. Estimated resin used
3. Best printing orientation (option to do this automatically)
4. Auto bed arrangement after supports added

Maeyanie, 2019-04-04:
I'd also love to see infill options instead of purely hollow, similar to FDM, to add rigidity and better support the top of the print. This won't save as much resin as it would filament, since some liquid is trapped inside when the top gets printed, but it still helps greatly with bed adhesion would save some, especially with taller prints.

bubnikv, 2019-04-05:
Do you have a suggestion of an SLA slicer that we may learn from? What
patterns are used in the industry?

Maeyanie, 2019-04-05:
The only ones I know are Formware, which does lattice infill, and NanoDLP, which does honeycomb. Most of them just offer hollow, or nothing.

The lattice is the better of the two since it leaves space for resin to drain, where the honeycomb captures it, but it's also more complicated to implement since it's not just the same pattern on each Z level.

I'd think many FDM infill patterns should work to some extent, though... rectilinear wouldn't work all that well except at high densities because of the bridging, and cubic wouldn't be great because it would be constantly trapping resin, but a lot of them should be fine.

bubnikv, 2019-04-05:
I found a screenshot from Formware, such an infill is quite simple to
generate. I installed NanoDLP slicer, but I did not find a way to generate
any infill structure. A screenshot would be welcome.

[image: image.png]

Maeyanie, 2019-04-06:
NanoDLP's UI is rather simplistic.
Config:
[image]
Results (25% size to not be huge):
[image]

bubnikv, 2019-04-06:
thanks. the lattice infill makes sense, while all the other fdm infills don't as they do not allow the resin to drain.

Maeyanie, 2019-04-06:
Yes, the lattice is ideal for exactly that reason, so if it's simple like you said it would be great to have... but even vertical cells use less resin and reduce film adhesion compared to solid fill, while being stronger and more supporting than completely hollow, so are better than nothing.

bubnikv, 2019-04-08:
I don't agree that vertical cells are a good idea. You don't want to face
the wrath of our customers having uncured resin entrapped inside these
cells and cracking the print, having the uncured resin spilling all over
the place.

Maeyanie, 2019-04-08:
In my experience the post-cure solidifies the trapped resin, so it hasn't been a problem. But I guess with thick walls, particularly opaque resins, or certain geometries it would be an issue, and the stuff generally isn't healthy to get everywhere, so that is a good point.

port513, 2019-05-17:
Still no option to hollow out a print in latest PrusaSlicer, is this a work in progress or do we have to use a second software for this?

bubnikv, 2019-05-17:
it is a work in progress and you have to use a 3rd party software for that.

port513, 2019-05-23:
> it is a work in progress and you have to use a 3rd party software for that.
> […](#)

I can see that but it will be there soon? SL1 is released and that is a really important piece for a good slicer.

Maeyanie, 2019-05-23:
It's fairly simple and quick to do in Meshmixer beforehand, so wouldn't say it's a huge problem to not have... it would be handy, though, for sure.

verner33ob, 2019-06-27:
Just found this post.
Industry standard is Magics Materialise.
Other software for inspiration would be Autodesk Netfabb, ChituSlicer, Unyk and Creation Workshop can also create Support. Right now i don't have time, but i extensively used and tested all these programs and know how they create there supports and what the pros and cons of these softwares are.
So in my head i have the perfect support generation software.
But yeah, i think, to write a list of everything i found out about those programs, will most definitely be a too big novel for a simple post on github :D

Maeyanie, 2019-06-27:
According to https://youtu.be/hZ3Dp7pwroA all Materialise Magics supports is different variations of "hollow." If that's the industry standard infill pattern, I'm not impressed.

verner33ob, 2019-06-28:
The video just shows the boolean operators and the hollowing. That has nothing to do with support generation. But those functions are damn good and fast in this program. Also repairing of models is.

Maeyanie, 2019-06-29:
I don't understand. This thread doesn't talk about supports, PrusaSlicer does fine with those already. Why would it need to have anything to do with them?

verner33ob, 2019-07-01:
I'm also sorry, i was confused and basically answered the question asked about other silcers and also one from another thread :D Sorry. But yes, these programs are also used for the inside supports. I just wanted to throw them out here for the developers here, to see what other programs do.

bubnikv, 2019-08-01:
A nice infill would be a complement structure to a gyroid infill as somebody suggested somewhere (sorry, I forgot the source).

> Hollow model with or without draining holes solid or open bottom on the model etc.

That is in the planning for PrusaSlicer 2.2 release.

> Best printing orientation (option to do this automatically)

We have some implementation, but it is not perfect.

> Auto bed arrangement after supports added

Do you need that? You should not care too much if the pads or the supports intersect, on the contrary, they will hold better.

> Estimated resin used
> We need to be able to add price for the resin so Slic3r PE can calculate the cost to produce the object too. Can't find that for resins at the moment.

That is quite an omission. We should fix that. @YuSanka would you please put it on your list? Thanks.

milkowski, 2019-09-06:
Gyroid infill seems the way to go since it allows resin to get out and also is structurally strong.
My request would be to have an option to add supports only in necessary places in auto-support option (where the cross-section plane has disjoint areas) since now support pillars appears on unnecessary places especially on high long objects that are structurally rigid by itself.
Also it would be great to support feature similar to Photon-S Resin Exposure Range Finder as described here https://www.thingiverse.com/thing:3818364 (with simple base simple base template like https://www.thingiverse.com/thing:3788554)

mylife4aiurr, 2019-10-13:
Why cant we hollow or make drain holes yet. 

+1 for this feature

xiteer, 2020-01-07:
I really appreciate all your hard work on PrusaSlicer!

+ 1 for:

- Resin Cost Calculations
- Hollowing and Drain Holes

lukasmatena, 2020-02-25:
Hollowing and the suggestions by @port513 are now part of 2.2.0-beta.

I think I will rename this issue and leave it open because of the discussion on SLA infill. Please, don't mix any other topics in here this time. Thanks.

AndrewFarley, 2020-05-01:
+1 on the lattice infill.  Would be great instead of the insane internal supports that currently are generating when hollowing.

r-a-i, 2020-05-03:
+1 on lattice infill. I am using it on the new version of ChituBox (they call it Grid3D) and the cross beam patterns print well, provide excellent drainage and structural stability.  It also looks pretty cool on clear prints.

Liger0, 2021-02-18:
Please add infill! The supports breaking in a model and hanging around are disturbing.

closed 2026-09-29 by prusaslicer-bot[bot]