prusa3d/PrusaSlicer #2056: Cut function needs option to leave items stacked not both on the table

joegchurch, 2019-04-01:
### Version
_Slic3r PE 1.41.3+ WIN64

### Window 10 Pro 10.0.17134 Build 17134

### Prusa MK3 MMU2 unmodified.
MK3 3.60 MMU2 1.05

### Behavior
* _When I cut a model it automatically arranges the two parts beside each other flat to the table. I then am unable to stack them back as two items as they were originally as one.  
When I import multiple stls at the same time I am asked if they are parts of the same model. 
When doing a cut It should do the same. If not giving a choice it should leave them stacked.
 This would make printing multicolor on the MMU2 easy.  
Now I have exported the two stls to Meshmixer, stacked them, then returned to Slic3r PE. Annoying and actually difficult to align the two parts in Meshmixer.

* _Steps needed to reproduce the problem_
Happens every time.

[image]

_Is this a new feature request?_
Yes or more of a correction for an existing feature

No file. Problem occurred before slicing shown.

Mike4U, 2019-04-01:
Thats what modifiers are for.  I was going to say read the instructions..  but now I can't seem to find where I learned how to do this!
Basically ins Slic3r with a MMU2 printer selected select your part and in the right menu select the box and select add modifier.  You will be able to move it around and resize it and select the extruder for it.

joegchurch, 2019-04-02:
Thank you.
Seems more obscure than it should. Even you don't remember how.
Could you not just make it an option when you cut?
Simple.
Make the mmu2 more valuable. At least to me.

joegchurch, 2019-04-02:
Now I am angry.
 So I should read some obscure documents because you chose to make Cut work
counterintuitively?
I expect my item to be in the same position after I cut as it was before.
Or at least be given a simple option to leave it so.
 So then I could slice it. In two colors. Easily. On my MK3MMU2. From you.
 Same position as before, like Meshmixer Plane Cut does. Like SketchUp
does. Likely like every other quality design program does.
You chose to move the upper piece elsewhere. With no way to put it back.
Moved without warning and without logic. And without choice.
What were you thinking?
And you defend this.
Instead of fixing it.
ColorPrint doesn't work well enough for MMU2. But at least it doesn't
reposition the results.
Slic3r PE should work better for MMU2. Not worse. It is your program. And
your product.
But only if you want MMU2 to succeed.
Did I mention I was angry.

tkircher, 2019-04-02:
Slic3r isn't a 3D modeling program. The purpose of a cut is to split an object into two pieces. Leaving one floating in the air makes no sense. If you want your model to be two different colors for MMU, use the correct tool.

joegchurch, 2019-04-02:
And what would that correct tool be?
Certainly not modifiers: "Modifier meshes are one of the most advanced
features of Slic3r."

joegchurch, 2019-04-02:
The purpose of a Cut to me is so that I can print on MK3MMU2 in two colors.
The top part would not be "floating in air" but be where it should be (and
was before) on top of the lower piece.
Cut should work just like when I use Add to load  a single object (.obj,
.stl....) having multiple parts.
Slic3r correctly pops up: "Multiple objects were loaded for a
multi-material printer. Instead of considering them as multiple objects,
should I consider these file to represent a single object having two parts?"
Works very well and so simple.
Too back Cut does not.
And you seem not to care about make the MMU2 work well.

bubnikv, 2019-04-02:
Hello @joegchurch. I am the lead developer of Slic3r PE. The two guys, @Mike4U @tkircher are just trying to help you, I suppose they are our customers or just happy users of the Slic3r PE on alternate machines. 

We certainly do care about our products, I hope it is clearly visible from the amount of work we are putting into the Slic3r PE development, just check the release logs and dates
https://github.com/prusa3d/slic3r/releases

Thank you for your opinion on how the cut tool should work. Please note that there are currently around 600 issues open, and each of the issue is important to someone.

Mike4U, 2019-04-02:
@bubnikv 
I was just trying to help.  I am a MK3 MMU2 owner.  Maybe this isn't the place for me to comment, especially since I'm possibly wrong.  Am I helping or hurting the situation?  I'm not mad or anything, just trying to decrease your work load.  Maybe I increased it?  If so sorry.

joegchurch, 2019-04-02:
I apologise. I thought I was interacting with someone at Prusa. My error.
I find it unnecessarily hard to do a simple simple Cut between layers in
Slic3r PE so that I can color, slice, export and save the resulting stls
positioned at the correct heights.
If I use Add with 2 or more stls at the same time Slic3r asks me if they
are parts one object. If yes they are put onto the table stacked (or side
by side or whatever), as they were originally in Meshmixer, Sketchup or
whence they came. If no, they are treated as multiple independent objects
with each arranged flat to the table. Perfect.
If I use Cut on an object,Slic3r PE makes the horizontal cut part easy. But
then without asking it takes the two pieces of what was previously one
object and arranges them as two unrelated objects each flat to the table.
There is no option with Cut as there is with  Add to choose multiple parts
of the same object (which clearly they were).
They should ask the same question as with Add.
Then I could Add a single part object, Cut it horizontally wherever I want,
choose multiple parts of the same object, choose the colors for each of the
parts, slice, export the g-code and print on my MK3MMU2 in multiple
colors.  No muss, no fuss.
I could also save the parts in their proper positions as separate stls.
Then I could reload the stl files again later as parts of a multipart
object, change the part colors and slice again. Likely I could Cut one or
more of the parts to increase the colors.
That would be perfect.
Too bad I (and you) can't.
I see no good reason why not.

bubnikv, 2019-04-03:
@Mike4U 
> was just trying to help.

We are always thankful when a community member helps another community member. This takes some of the burden off our shoulders, and we are left with more time to fix and improve the software.

bubnikv, 2019-04-13:
By the way, this issue is a duplicate of https://github.com/prusa3d/Slic3r/issues/1966
Therefore I am closing this issue, keeping https://github.com/prusa3d/Slic3r/issues/1966 open.

closed 2019-04-13 by bubnikv