prusa3d/PrusaSlicer #2057: Don't put seem into overhang/bridge if possible

espr14, 2019-04-01:
1.4.2 a7 win64

Perimeter has some parts supported and some are bridges and overhangs. Slic3r is set to random seem and it put a seem into a bridge. This will cause bridge to fall and the overhang part will need a lot more layers to build itself up.

[image]

tkircher, 2019-04-02:
How do you plan to print that hole in midair?

supermerill, 2019-04-02:
@tkircher It's an other issue.

espr14, 2019-04-02:
@tkircher There are many layers above. First few layers will fall down but then it will start to create correct shape. Try it and you'll see. It will be hidden in the box so quality and appearance do not matter. It will look like this:

[image]

espr14, 2019-04-03:
This is the result. Few hanging strings but the rest is perfect.

[image]

bubnikv, 2019-04-13:
Model? 3MF?

espr14, 2019-04-13:
In 1.4.2 b1 it does not create seem in the overhang.
[overhang hole.zip](https://github.com/prusa3d/Slic3r/files/3076420/overhang.hole.zip)