prusa3d/PrusaSlicer #2058: Wipe tower print speed

espr14, 2019-04-02:
Wipe tower can be ugly so it does not make sense to keep print speed low. Also acceleration can be quite high. However, there's no possibility to set speed and acceleration for wipe tower.

bubnikv, 2019-04-13:
for what printer? Prusa i3 MMU2? About which extrusion are you talking about? For MMU2, some of the print moves have a well defined speed defined by the ramming profile.

espr14, 2019-04-13:
Yes, Prusa i3 MMU2. How to achieve slow speed for the printed model and high speed for the wipe tower?

MR2C280, 2019-07-04:
Has anyone found out what controls the speed of the wipe tower printing?  For me, it's much too fast (at lease the first couple of layers).  It prints a line then immediately reverses.  If the plastic didn't attach to the tower wall, a little raised _wave_ is created.  Subsequent passes add to and pull up the _wave_ and, eventually, it hooks on the fan duct which drags it up and off the tower and you've got a right mess and no tower.  The only thing I could find was setting the max X print speed.

daavery, 2019-09-22:
same problem printing a regular filament and a flex , the flex needs a much lower speed everywhere to prevent extruder jams,

bubnikv, 2019-09-23:
In PrusaSlicer 2.1, the speed over the wipe tower is controlled by the
filament ramming control and the maximum filament volumetric speed
controls.

MR2C280, 2019-09-23:
> In PrusaSlicer 2.1, the speed over the wipe tower is controlled by the
filament ramming control and the maximum filament volumetric speed
controls.

The problem with using volumetric speed to slow down the wipe tower is that it slows the **whole** print down (for that filament).  The wipe tower slice doesn't seem to respect Max Print Speed (or any other parameter I tried like Acceleration or Infill/Perimeters Overlap).  I'm sure part of the problem must be the sudden reversal at the ends of the wipe tower and lack of overlap.
[image]
).

bubnikv, 2019-09-23:
What is the problem that you are trying to solve?

I suppose for @MR2C280 , the solution would be to slow down the wipe tower print speed at the first layer, which we most likely do not do as of now, and I understand the request.

IMHO the request by @espr14 to bump up the wipe tower extrusion speed to maximum has been solved by PrusaSlicer 2.1.0 by respecting the maximum filament volumetric speed.

espr14, 2019-09-24:
This issue is about maximizing print speed of wipe tower. Shape of wipe tower is being discussed in another issue.

bubnikv, 2019-09-24:
> @espr14  This issue is about maximizing print speed of wipe tower. Shape
of wipe tower is being discussed in another issue.

Then just set the filament volumetric speed to your maximum that you want
to extrude over the wipe tower.

espr14, 2019-12-21:
That's dangerous since this could be very high and you don't want to print external perimeter at the highest speed possible. If a tower wall collapses then the whole print could be thrown away.

benr916, 2020-06-03:
The Max Volumetric Speed setting in either the Filament Settings > Advanced section or the Print Settings > Speed section is ignored in the wipe tower. It only applies to the object being printed.
You can test this. Set the Max Volumetric Speed in both sections to say 8 mm3/s and slice an object. Then change the view (in the Plater tab, on the lower left corner) to Volumetric flow rate and youll see that the flow rate will stay under 8 mm3/s for the object but will go much higher in the wipe tower as though the slicer is ignoring the Max Volumetric Speed in the wipe tower.

tommylux, 2021-11-20:
How can I slow down the wipe tower, it prints too fact and causes it to become detached!

rtyr, 2021-11-21:
@tommylux Maximum print speed on the wipe tower is limited by max. volumetric speed in filament settings. There is also a new setting which allows you to specify brim width for wipe tower (available in 2.4.0-beta releases).

tommylux, 2022-03-02:
What is a suitable parameter range to limit the speed of the wipe tower to around 50 ms/s?

5jvm0u4, 2022-04-24:
> @bubnikv 
> Then just set the filament volumetric speed to your maximum that you want
to extrude over the wipe tower.

This does not work as described, my printing speed is kept at around 7mm^3/s while my max. vol. speed is set to be 20mm^3/s. (PrusaSlicer 2.4.1)

[image]

tommylux, 2022-11-26:
On first inspection, it looks like the 'Print Settings' - 'Max Volumetric Speed' was not doing anything. If your you have the 'Max Volumetric Speed' defined in 'Filament Settings', this appears to override the default value in 'Print settings'.

A better use of this feature may be a calculation of MAX between the two values, but now I know how this works, I'm happy to set the 'Filament Settings' to 0 when I want to use the default value.

Now I know how this works, when I set the 'Max Volumetric Speed' to 1 for example, I can see the model is reflected in the 'Volumetric Flow Rate', but the wipe tower remains at 15mm3/s
[image]

If I try and set a 'Max Print Speed' the wipe tower ignores these values, both speed and volumetric flow.
[image]

At this flow rate, the extruder skips when wiping. I use KFactors so there is increased acceleration on the extruders. I need to be able to limit the flow rate or control the speed on the wipe tower, but it ignores the limits set. I cannot see anywhere in the settings where it is finding 80mm/s for the wipe tower print speed.

[image]
[image]
[image]

How do I slow down the Wipe tower print speed?

tommylux, 2022-11-26:
Max Print Speed and Max Volumetric Speed in the 'Print Settings' only affects model, not Wipe Tower. I am unsure how the wipe tower can reach speeds up to 80 mm/s when this is not defined anywhere.

Adjusting the 'Max Volumetric Speed' in the 'Filament Settings' does affect the wipe tower.

aridShrub, 2022-11-27:
I believe acceleration setting affects wipe tower speed at least initially, based on the sound of it speeding up. The wipe tower is too small especially with support filament, but have seen very weak inter layer adhesion on Z coordinates with layer changing to Support W. 

From a print throughout perspective super fast printing of wipe tower is swamped out by layer changing time. I have also experienced failures due to poor printing of the wipe tower, lots of stringing and fouling of the nozzle, this debris eventually destroyed the print. Please add speed setting for wipe tower! I have found I can greatly improve print quality by tuning wipe tower parameters. If you look in the filament type list you can control everything listed until wipe tower.

Another missing feature is a draft shield, cura has a setting for this. [image][image]

aridShrub, 2022-11-27:
Ideally this draft shield is built with all active layers as per cura, in an interlock/interleaving pattern, or perhaps just use same number of perimeters as the number of layers being printed at the z height

bombela, 2022-12-14:
My max volumetric flow rate is 18mm³/s. But I only et ~8mm³/s on the wipe tower. I couldn't find any way to speed that up.

[image]

Note that I do not set the ramming/unload/load within prusaslicer. I set all values to zero, disabling it, and I filter out the left over `M900 K0` that prusaslicer leave behind for some reason (pressure advance setting specific to marlin). I am using klipper, and handling the ramming/unload/load with it directly.

lukasmatena, 2023-11-15:
Since [2.6.1-rc1](https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.6.1-rc1), the wipe tower speed is determined from perimeter speed and infill speed, as mentioned in the change log. Closing.

closed 2023-11-15 by lukasmatena

skoblins, 2024-10-07:
I only want to decrease the speed of the FIRST layer of the wipe tower, because it doesn't always stick well to the plate.

espr14, 2024-10-07:
> I only want to decrease the speed of the FIRST layer of the wipe tower, because it doesn't always stick well to the plate.

In case of the wipe tower, I would make extrusion width bigger. This sticks much better than decreasing the speed. But there's no setting for this (yet).