prusa3d/PrusaSlicer #2059: [Feature Request] Add option to choose extruder in Single Mode

dikodahan, 2019-04-02:
### Version
1.42.0-Beta1

### Operating system type + version
All OS

### 3D printer brand / version + firmware version (if known)
MK3S (FW v3.7.0) with MMU2S (FW v1.0.5)

### Behavior
When printing a multi-color print you have the option to designate for each part which extruder to use:
[image]

This does not exist is you are printing in Single Mode / Single color. 
[image]

I am suggesting adding the exact same option in single mode printing with MMU2S.

bubnikv, 2019-04-03:
why dont you just use the mm printer profile then?

dikodahan, 2019-04-03:
I am aware of the ability to create multiple individual profiles per extruder.
But, since you already have this functionality available, from a UI/UX perspective this would be a minor change (coding wise) but would make usability much simpler and easier!

bubnikv, 2019-04-13:
From the UI perspective maybe, but the full implementation would be much more involved.

rtyr, 2021-01-26:
MK3S MMU2S profile can be used for single prints as well, so it is already easily possible to do what you want.

It doesn't make sense to implement such option for single material printers or specifically to "MK3S MMU2S Single" printer profile (which is single material printer from the PrusaSlicer PoV), this profile will trigger filament selection on the printer anyway.

Closing.

closed 2021-01-26 by rtyr