prusa3d/PrusaSlicer #2060: Problems slicing text

tiendaaep, 2019-04-02:
### Version
_Version of Slic3r Prusa Edition used goes here_1.42.0

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version Windows 10 
_What OS are you using, and state any version #s_
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known) MK3
_What 3D printer brand / version are you printing on, is it a stock model or did you modify the printer, what firmware is running on your printer, version of the firmware #s_

### Behavior
* _Describe the problem_
I slice some pieces with text. When i slice with Slic3r PE some parts of the text does not show and of course does not print. The problem is in Slic3r PE. If i use Cura hte text is complete. 
  * _Screenshots from __*Slic3r*__ preview are preferred_
[image]

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

bubnikv, 2019-04-03:
would you attach the project file and screenshots of cura? Will enabling
the thin walls help?

tiendaaep, 2019-04-03:
Hi Bubnikc, i have enabled the thin walls and the slicing of the text has
worked ok.

Thank you.
Best regards.
Isidoro.

tiendaaep, 2019-06-20:
Hi Bubnikc, i have problems slicing my works.

When i must print text there is some lines that does not appear after
slicing. I have text 3mm high.

What could i do to avoid this and print it ok?.

Thank you.
Best regards.
Isidoro.

tiendaaep, 2019-06-21:
Hi Bubnick, As i said in other occasion when i slice the model with CURA it
works ok.

Enclosed i send the screenshots of CURA and the RMF of the model.

Thank you.
Best regards.
Isidoro.

closed 2026-09-29 by prusaslicer-bot[bot]