prusa3d/PrusaSlicer #2061: Highly inaccurate print time (Slic3r 11h 19m, reality 15h50m)

szymonpk, 2019-04-03:
### Version
1.41.1

### Operating system type + version
Operating System:    linux
System Architecture: x86_64-linux-thread-multi
System Version: Linux nefele 5.0.0-gentoo #3 SMP Sat Mar 9 10:18:53 CET 2019 x86_64 Intel(R) Core(TM) i7-8650U CPU @ 1.90GHz GenuineIntel GNU/Linux

### 3D printer brand / version + firmware version (if known)
Prusa MK3S, 3.6.0

### Behavior
I should also mention I am using latest, stable OctoPrint (1.3.10) and I didn't noticed such discrepancies with other prints.

Slic3r estimated print time for attached STL to 11h19m in reality it was printing almost 16h. See attached G-code and STL. I am not sure if it is printers fault of OctoPrint?

Print finished perfectly fine. Just a tad too long ;-)

#### Project File (.3MF) where problem occurs
- [G-code](https://www.dropbox.com/s/fswyq3n5yri5tle/fume_extractor_head_PET_0.2x0.4_240-235_75-80.gcode?dl=0)
- [STL](https://www.dropbox.com/s/emx8teofqegd83q/fume_extractor_head.STL?dl=0)

bubnikv, 2019-04-03:
It may very well be, that the g-code is too dense to be processed at
full speed if printed with octoprint. The most reliable test is to print
the gcode from an sd card. Even then it may happen that the gcode is too
detailed for the printer to keep up.

szymonpk, 2019-04-03:
Ok :( I didn't know G-code density may be the issue. If it is expected please close this issue. Unfortunately, I don't have plans to print this part second time (from sd card).

szymonpk, 2019-04-03:
I just wanted to mention that OctoPrint host isn't loaded with anything else. System load is low (5 minute load < 0.5) and CPU usage is below 30% all the time. Although I might have slow sd-card there, but this is only 20MB of G-code.

bubnikv, 2019-04-04:
I have asked @jindrichbenes to print it from an SD card to verify the print time estimate.

jindrichbenes, 2019-04-05:
We ran the print at two random MK3S and the print shows exact 11H 19M.

[image]
[image]

szymonpk, 2019-04-05:
Thanks for checking this out! I don't have issues with other prints big or small. I assume it must be one of two things mentioned by @bubnikv. I don't think there is much more to discuss here. Again, thanks for awesome support, and sorry I have doubted your theory.

closed 2019-04-05 by szymonpk