prusa3d/PrusaSlicer #2062: 1.42.x completely unusable in debian with cinnamon

prusa3duser, 2019-04-03:
### Version
1.42 (any of them)

### Operating system type + version
Debian. Fully up to date. Running cinnamon.

### 3D printer brand / version + firmware version (if known)
Prusa MK2S

### Behavior
The 1.42.x versions of the slic3r PE are completely unusable in Debian running cinnamon.  The interface is extremely slow.  The 1.41.3 version does just fine.

Even just bringing up the GUI and moving around within it, without loading a project nor STL is slow as molasses!

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

bubnikv, 2019-04-04:
the 1.42 depends on opengl shaders, while 1.41 and older would work in
opengl 1.1 graphics cards. If your opengl is software rendered, then the
1.42 renders much slower than 1.41. What opengl version does Slic3r report
in Help -> System info?

prusa3duser, 2019-04-05:
Hi thanks for the quick response!
Help->System Info reports this:
     **GL Version: 3.0 Mesa 13.0.6**

prusa3duser, 2019-04-05:
Thank you! I installed the proprietary nvidia drivers for my video card and now it works just fine. You can close this issue. :+1: 
Help->System now reports:
**GL Version: 3.3.0 NVIDIA 340.106**

closed 2019-04-05 by prusa3duser