prusa3d/PrusaSlicer #2063: Slic3r 1.42.0 cuts away part of the model

Rick-T, 2019-04-03:
### Version
I built Slic3r from the latest commit on the master branch (as of writing this it is [703b9bda324d656182e17cdf95b82abf358819e6](https://github.com/prusa3d/Slic3r/commit/703b9bda324d656182e17cdf95b82abf358819e6).

Although the problem also occurs on the Slic3r 1.42.0-beta1. 

It does **not** happen in 1.41.2.

### Operating system type + version
Linux

### 3D printer brand / version + firmware version (if known)
Prusa MK3 MMU2

### Behavior
Slic3r cuts away part of the model. I am using stock profiles and settings for everything.

You can see the problem in the attached screenshots. Part of the tail is missing. I have observed this only in Slic3r 1.42.0. In 1.41.2 it slices fine. Cura is also able to slice the model correctly.

[image]
[image]

I am using [this beautiful model from thingiverse](https://www.thingiverse.com/thing:2931434). The problem occurs on the first file for the multi color object.
#### Project File (.3MF) where problem occurs
[eevee_lowpoly_flowalistik_dual1.3mf.gz](https://github.com/prusa3d/Slic3r/files/3040158/eevee_lowpoly_flowalistik_dual1.3mf.gz)

bubnikv, 2019-04-04:
I think you loaded just a single part of a two part object by mistake.

Rick-T, 2019-04-04:
@bubnikv: Thank you for you response. Please note that I intentionally loaded just the single part to show the error. If I were to load the whole model the missing part of the tail would still be missing. I know this, because I printed the whole model and it came out with a missing section.

Please take a look at the screenshots again. The tail in the model (first screenshot) ends with a spiky corner, the sliced tail (second screenshot) ends flat. That has nothing to do with the object being part of a multi-part model.

I you inquire I can post more screenshots and a photo from my print with the missing section once I am home from work. But please consider checking this out in the Slic3r yourself. 

Also note, that I edited my OP. The problem only occurs Slic3r is built from the master-Branch. 1.42.0-beta1 and 1.41.2 work fine, so the error has to be caused by one of the more recent commits.

bubnikv, 2019-04-04:
I see, it is this tip of the tail that is missing.

[image]

Slic3r shows that there are errors in the model. If you click on the highlighted symbol with the right button, on Windows 10 the model is sent to the Netfabb fixing service. Once returned, it slices correctly. Unfortunately the service is only integrated on Windows 10 as Microsoft provides access to it through Windows API.

[image]

We will keep this issue open and we will review the model fixing issues in Slic3r after more data is collected.

bubnikv, 2019-04-04:
I should be reading your comments more carefully. Thanks for a detailed and a very useful report indeed. I will review the changes between the beta1 and the current master.

bubnikv, 2019-04-04:
This commit broke it 382326ffc85620edbe320e956439caccca9df271 , actually the optimization I did in the first function.

bubnikv, 2019-04-04:
Fixed with ef1a273f0c31e6f606681a9b94be41c3d09d5feb
Thanks for the report.

closed 2019-04-04 by bubnikv