prusa3d/PrusaSlicer #2065: Gyroid infill has a really high resolution

SijmenSchoon, 2019-04-04:
### Version
8736bb46

### Operating system type + version
Arch Linux

### 3D printer brand / version + firmware version (if known)
Original Prusa i3 MK3S (with OctoPrint on RPI 3B+)

### Behavior
When printing over USB (in my case with OctoPrint on a Raspberry Pi 3B+), the printer seems to slow down (and stutter) when printing infill compared to my other printer (which has a DuetWiFi).

This seems to be due to the gyroid infill using a really high resolution, at one point per 0.1mm (if I understand the [relevant code](https://github.com/prusa3d/Slic3r/blob/master/src/libslic3r/Fill/FillGyroid.cpp#L70) correctly). This works out to around 50 points per wa

ve on 15% infill, or 1000 points per second when printing at 100 mm/s, which seems a bit excessive to me.

I suggest increasing the tolerance (to 0.5 mm or something), or making it configurable. In the gcode preview it seems to still look fine, but it reduces the points per second to about 200 - much better. Also, it reduced the gcode size of my test file from 2.9MB to 1.8MB.

I've yet to test this change on OctoPrint - I will do that this evening.

_Is this a new feature request?_ No

#### Project File (.3MF) where problem occurs
Should not be relevant, but [Helical_Cage_Gear_Base_v2.zip](https://github.com/prusa3d/Slic3r/files/3042955/Helical_Cage_Gear_Base_v2.zip)

bubnikv, 2019-04-04:
I would be worried, that just reducing the resolution in each of the layers
independently from the other may lead to unreliable overlap of successive
layers.
Ideally the gyroid infill would be discretized by the printer firmware.

SijmenSchoon, 2019-04-04:
Perhaps the [arc movements](http://marlinfw.org/docs/gcode/G002-G003.html) could be of use?

bubnikv, 2019-04-05:
maybe, but not all firmwares implement the arc movement G-code. And the arc
movement is just a rough approximation of the gyroid.

Maybe we could add a gyroid accuracy parameter, but that would add another
one to the list making it a mess like the Cura, and it would need to be
tested throughly.

wavexx, 2019-08-21:
This should be fixed by #2730 in combination with newer firmwares

wavexx, 2021-01-26:
I think this should be effectively fixed now.

closed 2021-01-26 by rtyr