prusa3d/PrusaSlicer #2066: Missing layers after scale to 334%

JohnBrods, 2019-04-04:
### Version
_Version of Slic3r Prusa Edition used goes here_ 1.41.3 win64 bit

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version 
_What OS are you using, and state any version #s_ Windows10 64 bit
_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### 3D printer brand / version + firmware version (if known) Prusa i3MK3S latest firmware

### Behavior

I cut a model from thingiverse into 4 parts with Mesh mixer, scaled it to 334% and the Slic3r is missing parts of the roof, many layers are missing. Seems ok if the stl is not scaled up.

Import stl, scale to 200% or 334 and preview.
 
[image]

_Is this a new feature request?_No

I used Cura to slice, that works ok except I always get lines of filament in the wrong place where there should be nothing, see picture below for cura slice actual print.

[image]

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

JohnBrods, 2019-04-04:
I just scaled the model in Meshmixer to 334% before exporting, Slic3r is ok with the stl but is unable to scale and slice by itself 100% of the time. I've had other models with missing layers. Also, I printed the Washington Monument, the top triangular roof should have been flat, Prusa Sli3r 3 added a spiral roof. That has happened on the castle stl file that Prusa supply with their new printers.

bubnikv, 2019-04-13:
Please provide the model.

JohnBrods, 2019-04-13:
That's just it, this site would not allow me to upload the .3MF project file. 
As I mentioned, I used meshmixer to cut a large model, Library Of Congress, into 4 quarters. 
I then scaled them up using Slic3r and had the issue.
I eventually scaled them up in meshmixer then sliced in Slic3r with no issue.
[image]

closed 2019-04-13 by JohnBrods

bubnikv, 2019-04-13:
just zip the 3mf file and then you can upload it.