prusa3d/PrusaSlicer #2067: [Feature Request] Octoprint upload to sd

LeonB, 2019-04-04:
At the moment when a file is sent to Octoprint it will always be placed in the 'local' location of Octoprint:

> location – The target location to which to upload the file. Currently only local and sdcard are supported here, with local referring to OctoPrint’s uploads folder and sdcard referring to the printer’s SD card. If an upload targets the SD card, it will also be stored locally first.

It would be nice for this to be configurable so you would be able to upload to the SD file via Octoprint. Either in the dialog after clicking on 'Send Gcode' or in the printer settings pane. 'Start printing after upload' would also have to account for this.

What is the more preferable location for this configuration option? Settings pane or upload dialog? And would a pull request for this be accepted? Or is this a feature Prusa is not interested in having in Slic3rPE?

wabbitguy, 2019-04-08:
Uploading to the SD card via Octoprint puts new meaning to the word "slow". Probably faster to order the print from a printing service...

https://community.octoprint.org/t/upload-to-sd-takes-too-long/231

marcosscriven, 2021-05-24:
@josefprusa suggested a while back it would be possible to improve the speed of this https://twitter.com/josefprusa/status/915315958545289222

Is it possible to get a comment from Prusa on this please?

bubnikv, 2021-05-24:
> Is it possible to get a comment from Prusa on this please?

This is a firmware topic I suppose.

https://github.com/prusa3d/prusa-firmware

marcosscriven, 2021-05-24:
@bubnikv - I added there now, thanks https://github.com/prusa3d/Prusa-Firmware/issues/3159

KjellVerb, 2023-01-14:
Is there any progress on the original request? I too like the peace of mind that SD-printing provides compared to octoprint serial printing.

closed 2026-09-29 by prusaslicer-bot[bot]