prusa3d/PrusaSlicer #2068: Buttons replaced by blue boxes.

Hotpocketdeath, 2019-04-04:
### Version
1.42 Beta

### Operating system type + version
Windows 10 Pro 1809

### 3D printer brand / version + firmware version (if known)
Prusa MK3

### Behavior
Buttons are missing.  Replaced with blue boxes

I run Slic3r on a remote desktop.  So I use RDP to connect to the machine.  This behavior was not present in previous version.

[image]

bubnikv, 2019-04-04:
Remote Desktop on Windows only provides OpenGL 1.1 through a software
rasterizer. Current slic3r requires OpenGL 2.0. Slic3r should detect that
Windows only offers OpenGL 1.1, in that case it should unload the defaut
OpenGL library and load the MESA software renderer library, which we bundle
with Slic3r.

Did you restart Slic3r after connecting over a remote desktop? Does that
help?

bubnikv, 2019-04-04:
If that does not help, you may try to copy the opengl.dll from the Mesa
directory one level up, but then you will always get the software renderer,
which is quite slow.

Hotpocketdeath, 2019-04-04:
Restarting while in RDP doesn't help.  They always come up as blue boxes.  If it's supposed to be loading the software renderer, then it sounds like it isn't doing it.

Copying the Mesa OGL down does help though.  But I wouldn't want to be stuck that way.  Slic3r is on my home desktop and I connect to it from work to run jobs.  But I do occasionally work in slic3r from the computer itself so it would be nice to have full use of it.

bubnikv, 2019-04-05:
What operating system is running on the server (where slic3r is running) and what on the client (where the RDP client is running)?

I tested the RDP access while ago with Windows 7 on server and Windows 10 on clinet. The issue may be, that the default OpenGL driver does not unload cleanly. The two solutions from the top of my head:

1) Start Slic3r with a command line parameter to use software renderer, if executed over RDP.
2) Use some Win32 API to detect a RDP connection and load the software renderer. This is likely possible, but it may require some googling and head scratching.

bubnikv, 2019-04-05:
We can likely detect RDP client if there is on graphics acceleration enabled on the server
https://docs.microsoft.com/en-us/windows/desktop/termserv/detecting-the-terminal-services-environment

You likely may try to enable graphics acceleration on your Windows 10 box acting as an RDP server (maybe just on Windows 10 enterprise, I did not test it)

https://docs.microsoft.com/en-us/windows-server/remote/remote-desktop-services/rds-remotefx-vgpu

bubnikv, 2019-04-05:
I have just tested connecting to both Windows 10 and Windows 7 using RDP and starting Slic3r there, and the Slic3r correctly loads the software renderer. I will add an explicit check for the RDP connection anyway just to be on the safe side.

bubnikv, 2019-04-05:
Would you please test this development build? The slic3r.exe wrapper tests for the remote desktop flag, and in that case it loads the MESA software renderer without hesitation.

[Slic3rPE-1.42.0-beta1+171-win64-full-ga9223aeb5-201904050813.zip](https://github.com/prusa3d/Slic3r/files/3047154/Slic3rPE-1.42.0-beta1%2B171-win64-full-ga9223aeb5-201904050813.zip)

Hotpocketdeath, 2019-04-05:
Wow, thank you.  That new version worked immediately.  I know it was probably a minor issue and not many people probably use Slic3r the way I do, but it's much appreciated to see a fix so quickly.

And sorry for not responding to some of your earlier suggestions.  I just woke up a couple hours ago and just got to work.

bubnikv, 2019-04-05:
No problem. Thanks for the report. I use RDP myself a lot for various purposes, so it surprises me that it did not work on your computer as it worked on two of mine.

Closing.

closed 2019-04-05 by bubnikv

chuckhays, 2019-05-29:
I see this same issue without remote desktop, on Windows 10 build 1803.

regnull, 2019-06-15:
Same issue here, I have completely uninstalled all Prusa software, deleted the directories under Program files and reinstalled from scratch, this did not fix the issue. Also tried the development build, the same issue.