prusa3d/PrusaSlicer #2069: slic3r-gui crashes on startup

Rick-T, 2019-04-04:
### Version
Commit [ccee04cd689017ba4880e6cd74d8c170ac40e7af](https://github.com/prusa3d/Slic3r/commit/ccee04cd689017ba4880e6cd74d8c170ac40e7af)

### Operating system type + version
**Operating System**: Unix
**System Architecture**: 64 bit
**System Version**: Linux 4.20.17-1-MANJARO x86_64

### OpenGL installation
**GL version**: 4.6.0 NVIDIA 418.43
**Vendor**: NVIDIA Corporation
**Renderer**: GeForce GTX 970/PCIe/SSE2
**GLSL version**: 4.60 NVIDIA
<details>
<summary>Installed extensions</summary>
GL_AMD_multi_draw_indirect
GL_AMD_seamless_cubemap_per_texture
GL_AMD_vertex_shader_layer
GL_AMD_vertex_shader_viewport_index
GL_ARB_ES2_compatibility
GL_ARB_ES3_1_compatibility
GL_ARB_ES3_2_compatibility
GL_ARB_ES3_compatibility
GL_ARB_arrays_of_arrays
GL_ARB_base_instance
GL_ARB_bindless_texture
GL_ARB_blend_func_extended
GL_ARB_buffer_storage
GL_ARB_clear_buffer_object
GL_ARB_clear_texture
GL_ARB_clip_control
GL_ARB_color_buffer_float
GL_ARB_compatibility
GL_ARB_compressed_texture_pixel_storage
GL_ARB_compute_shader
GL_ARB_compute_variable_group_size
GL_ARB_conditional_render_inverted
GL_ARB_conservative_depth
GL_ARB_copy_buffer
GL_ARB_copy_image
GL_ARB_cull_distance
GL_ARB_debug_output
GL_ARB_depth_buffer_float
GL_ARB_depth_clamp
GL_ARB_depth_texture
GL_ARB_derivative_control
GL_ARB_direct_state_access
GL_ARB_draw_buffers
GL_ARB_draw_buffers_blend
GL_ARB_draw_elements_base_vertex
GL_ARB_draw_indirect
GL_ARB_draw_instanced
GL_ARB_enhanced_layouts
GL_ARB_explicit_attrib_location
GL_ARB_explicit_uniform_location
GL_ARB_fragment_coord_conventions
GL_ARB_fragment_layer_viewport
GL_ARB_fragment_program
GL_ARB_fragment_program_shadow
GL_ARB_fragment_shader
GL_ARB_fragment_shader_interlock
GL_ARB_framebuffer_no_attachments
GL_ARB_framebuffer_object
GL_ARB_framebuffer_sRGB
GL_ARB_geometry_shader4
GL_ARB_get_program_binary
GL_ARB_get_texture_sub_image
GL_ARB_gl_spirv
GL_ARB_gpu_shader5
GL_ARB_gpu_shader_fp64
[... 323 lines cut ...]
Prusa MK3 MMU 2

### Behavior
slic3r-gui immediately crashes upon startup. I don't even see any window. Just an error box with a stacktrace.

```
ASSERT INFO:
./src/generic/imaglist.cpp(66): assert "(bitmap.GetWidth() >= m_width && bitmap.GetHeight() == m_height) || (m_width == 0 && m_height == 0)" failed in Add(): invalid bitmap size in wxImageList: this might work on this platform but definitely won't under Windows.

BACKTRACE:
[1] wxGenericImageList::Add(wxBitmap const&)
[2] Slic3r::GUI::Tab::add_options_page(wxString const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)
[3] Slic3r::GUI::TabPrint::build()
[4] Slic3r::GUI::Tab::create_preset_tab()
[5] Slic3r::GUI::MainFrame::add_created_tab(Slic3r::GUI::Tab*)
[6] Slic3r::GUI::MainFrame::create_preset_tabs()
[7] Slic3r::GUI::MainFrame::init_tabpanel()
[8] Slic3r::GUI::MainFrame::MainFrame()
[9] Slic3r::GUI::GUI_App::OnInit()
[10] wxEntry(int&, wchar_t**)
[11] Slic3r::CLI::run(int, char**)
[12] main
[13] __libc_start_main
[14] _start
```

bubnikv, 2019-04-05:
How did you get the Slic3r system info if the slic3r crashes on start up? From another build of Slic3r?

This is our appimage from the master. Does it run for you?

[Slic3rPE-1.42.0-beta1+167-linux64-full-gccee04c-201904041615.AppImage.zip](https://github.com/prusa3d/Slic3r/files/3046770/Slic3rPE-1.42.0-beta1%2B167-linux64-full-gccee04c-201904041615.AppImage.zip)

It may be, that the current master does not test for the existence of the images, and if your install is broken, it will not complain, but crash.

Rick-T, 2019-04-05:
I used the AppImage from the 1.42.0-beta1 release to get the system info. That one still works for me.

I cannot try running your AppImage on my desktop right now, because I am not at home. I was able to run it on my work notebook. However that one has Windows installed (😞) and I'm using wsl to run Linux apps, so the whole setup is somewhat weird.

I know another user who experiences the same crashes as me on a 64bit Arch Linux system. Maybe I can get her to run your AppImage, otherwise I will try it myself tonight when I'm home again.

Rick-T, 2019-04-05:
Additional info: I am installing Slic3r through [this AUR package](https://aur.archlinux.org/packages/slic3r-prusa3d-git/). Maybe some changes in Slic3r caused that package to break. However the package does not do anything too funky for building the application so I don't really expect the package to be faulty.

bubnikv, 2019-04-05:
I have added a code for loading svg icons if available, and fall back to pngs if svgs are not available. With that change an error reporting for missing images was broken.

@vojtechkral will add diagnostics to missing images. I suppose that will reveal something.

vansoest, 2019-04-05:
> I know another user who experiences the same crashes as me on a 64bit Arch Linux system. 

Wink. :)

> This is our appimage from the master. Does it run for you?

@bubnikv The AppImage starts without any problems on my Arch Linux.

wavexx, 2019-04-05:
Same stuff here. Rebuild slic3r from master (commit d31590f438549440e87ed8bd46bd72e5f2317db1).

````
ASSERT INFO:
./src/generic/imaglist.cpp(85): assert "bitmapSize == m_size" failed in Add(): All bitmaps in wxImageList must have the same size

BACKTRACE:
[1] wxGenericImageList::Add(wxBitmap const&)
[2] Slic3r::GUI::Tab::add_options_page(wxString const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, bool)
[3] Slic3r::GUI::TabPrint::build()
[4] Slic3r::GUI::Tab::create_preset_tab()
[5] Slic3r::GUI::MainFrame::add_created_tab(Slic3r::GUI::Tab*)
[6] Slic3r::GUI::MainFrame::create_preset_tabs()
[7] Slic3r::GUI::MainFrame::init_tabpanel()
[8] Slic3r::GUI::MainFrame::MainFrame()
[9] Slic3r::GUI::GUI_App::OnInit()
[10] wxEntry(int&, wchar_t**)
[11] Slic3r::CLI::run(int, char**)
[12] main
[13] __libc_start_main
[14] _start
````

This is on debian unstable, wx 3.1.2.

Rick-T, 2019-04-05:
I just want to confirm that the AppImage works for me aswell.

wavexx, 2019-04-09:
Rebuilding today I still get the crash on startup. It seems that some effort was done to fix this, but the code looks broken to me:

https://github.com/prusa3d/Slic3r/blob/d8cb1b07921a816fa94cb259eaa645ff09d6a587/src/slic3r/GUI/wxExtensions.cpp#L444

When load_scaled_bitmap fails, *bmp is not initialized. We call the wxBitmap copy constructor with a null and get a crash when the icon is missing.

Looking at one call above load_scaled_bitmap, I get even more confused:

https://github.com/prusa3d/Slic3r/blob/d8cb1b07921a816fa94cb259eaa645ff09d6a587/src/slic3r/GUI/wxExtensions.cpp#L429

I assume scale_base should be used to initialize width or height here? Never used. And width/height are still set to zero. This cannot work (and it doesn't).

vojtechkral, 2019-04-09:
> When load_scaled_bitmap fails, *bmp is not initialized. We call the wxBitmap copy constructor with a null and get a crash when the icon is missing.

@wavexx I've got a branch where the error handling is fixed (along with other refactoring).

> I assume scale_base should be used to initialize width or height here? Never used.

It is used - it's a reference. (That said, I didn't write the code and I'm not a huge fan of the way png dimensions are hardcoded in the code right now...)

bubnikv, 2019-04-13:
Another beta is to be released today or tomorrow. Please let me know whether the issue has been fixed.

wavexx, 2019-04-13:
On Sat, Apr 13 2019, bubnikv wrote:
> Another beta is to be released today or tomorrow. Please let me know
> whether the issue has been fixed.

Doesn't crash anymore, but the startup assertion is still present as
before.

I noticed by chance now it's resolution-dependent: no error when I only
use the built-in laptop screen (again, this is on x11, wx3.1).

Also, there's now an assertion about missing icons which will make
slic3r-gui stop. It's ok, but I would synthesize a blank icon and write
a warning instead of asserting.

[btw, I really like the new icons]

wavexx, 2019-04-13:
The above PR fixes this for me. This is another scenario related to #1906 IMHO, where the inconsistent usage of scaling factors and ifdefs around retina code really makes things hard to maintain.

wavexx, 2019-05-10:
Even with #1906, commit 077321b228814fbca66937812ea204842582f4b2 makes the assertion re-appear (didn't have time to look where the size is miscalculated, but it's still in Tab.cpp).

bubnikv, 2019-08-07:
@wavexx the code is in continuous development. Do you still get the assert with our current master?

wavexx, 2019-08-07:
Not since SVG was switched as default. I think we can close this.

bubnikv, 2019-08-08:
thanks for heads up.

closed 2019-08-08 by bubnikv