prusa3d/PrusaSlicer #2070: Layer misalignemnt

guestisp, 2019-04-04:
### Version
Slic3r Prusa Edition 1.42.0-beta1+linux64

I'm fighiting with a strange issue. I've tried almost all slicer out there: Slic3r PE, Cura, Simplify3D.
Only with Slic3r PE, i'm getting very bad prints, like the following benchy. Some layers, seems to be thicker than others. It's like a wobble, but a real wobble would be present with all slicers (it's an hardware issue). Seems a layer shift, but that would be present on all slicers and on both sides of the print. In this case, I only have bad layers with Slic3r and, most of the time, only on one side of the print.

This particular benchy has the same bad layers in the same position every time I print.

Slic3r:
[image]

Simplify 3D:
[image]

Cura:
[image]

The printer (a Geeetech A20) with FYSETC dual-gear extruder, E3D v6 and capricorn tubing, is calibrated, the thin wall test is showing that the extrusion is exactly what I've set in Slic3r (0.40 or 0.45, it depends, but the results are the same), steps/mm are perfect (the extruder extrude exactly 100mm of filament, X, Y, Z moves exactly 100mm on each axis)

As you can see from the posted images, the Slic3r print is the worse one, with thicker layers.

I've tried multiple filaments, always show the same exact issue in the same exact place.
I'm thinking there is something strange with Slic3r when slicing the models.

I was reminded, right now when writing, that with the first beta or with the Alpha, i didn't have so many issues. Could be related to the new beta? I have to print something with the previous one....

Additionally, even in this case only with slic3r, I have a huge oozing and stringing, when printng multiple objects. The same multiple objects print with S3D, same retraction setting, doesn't show any issues.

I did about 30-40 retraction test, all perfect, even with Slic3r. The issue is only when printing something that is not a test (like this benchy, or Adalinda the Dragon from thingiverse and so on)

Any advice ? The following is my current configuration.

```
[print:Standard 0.20mm]
avoid_crossing_perimeters = 1
bottom_fill_pattern = rectilinear
bottom_solid_layers = 3
bridge_acceleration = 0
bridge_angle = 0
bridge_flow_ratio = 1
bridge_speed = 60
brim_width = 0
clip_multipart_objects = 0
compatible_printers = 
compatible_printers_condition = 
complete_objects = 0
default_acceleration = 0
dont_support_bridges = 1
elefant_foot_compensation = 0
ensure_vertical_shell_thickness = 0
external_perimeter_extrusion_width = 0
external_perimeter_speed = 40%
external_perimeters_first = 0
extra_perimeters = 1
extruder_clearance_height = 20
extruder_clearance_radius = 20
extrusion_width = 0.4
fill_angle = 45
fill_density = 10%
fill_pattern = rectilinear
first_layer_acceleration = 0
first_layer_extrusion_width = 0
first_layer_height = 0.3
first_layer_speed = 50%
gap_fill_speed = 20
gcode_comments = 0
gcode_label_objects = 1
infill_acceleration = 0
infill_every_layers = 1
infill_extruder = 1
infill_extrusion_width = 0
infill_first = 0
infill_only_where_needed = 0
infill_overlap = 10%
infill_speed = 60
inherits = 
interface_shells = 0
layer_height = 0.2
max_print_speed = 80
max_volumetric_speed = 0
min_skirt_length = 50
notes = 
only_retract_when_crossing_perimeters = 1
ooze_prevention = 0
output_filename_format = [input_filename_base].gcode
overhangs = 0
perimeter_acceleration = 0
perimeter_extruder = 1
perimeter_extrusion_width = 0
perimeter_speed = 60
perimeters = 3
post_process = 
print_settings_id = 
raft_layers = 0
resolution = 0
seam_position = aligned
single_extruder_multi_material_priming = 1
skirt_distance = 20
skirt_height = 1
skirts = 1
slice_closing_radius = 0.049
small_perimeter_speed = 15
solid_infill_below_area = 70
solid_infill_every_layers = 0
solid_infill_extruder = 1
solid_infill_extrusion_width = 0
solid_infill_speed = 100%
spiral_vase = 0
standby_temperature_delta = -5
support_material = 0
support_material_angle = 0
support_material_auto = 1
support_material_buildplate_only = 1
support_material_contact_distance = 0.2
support_material_enforce_layers = 0
support_material_extruder = 1
support_material_extrusion_width = 0
support_material_interface_contact_loops = 0
support_material_interface_extruder = 1
support_material_interface_layers = 3
support_material_interface_spacing = 0
support_material_interface_speed = 100%
support_material_pattern = rectilinear
support_material_spacing = 2.5
support_material_speed = 60
support_material_synchronize_layers = 0
support_material_threshold = 0
support_material_with_sheath = 1
support_material_xy_spacing = 50%
thin_walls = 1
threads = 4
top_fill_pattern = rectilinear
top_infill_extrusion_width = 0
top_solid_infill_speed = 25%
top_solid_layers = 3
travel_speed = 80
wipe_tower = 0
wipe_tower_bridging = 10
wipe_tower_rotation_angle = 0
wipe_tower_width = 60
wipe_tower_x = 180
wipe_tower_y = 140
xy_size_compensation = 0

[filament:AmazonBasics PLA Viola]
bed_temperature = 65
bridge_fan_speed = 100
compatible_printers = 
compatible_printers_condition = 
compatible_prints = 
compatible_prints_condition = 
cooling = 0
disable_fan_first_layers = 2
end_filament_gcode = "; Filament-specific end gcode \n;END gcode for filament\n"
extrusion_multiplier = 1
fan_always_on = 1
fan_below_layer_time = 60
filament_colour = #C900FF
filament_cooling_final_speed = 3.4
filament_cooling_initial_speed = 2.2
filament_cooling_moves = 4
filament_cost = 19.9
filament_density = 1.24
filament_diameter = 1.75
filament_load_time = 0
filament_loading_speed = 28
filament_loading_speed_start = 3
filament_max_volumetric_speed = 0
filament_minimal_purge_on_wipe_tower = 15
filament_notes = ""
filament_ramming_parameters = "120 100 6.6 6.8 7.2 7.6 7.9 8.2 8.7 9.4 9.9 10.0| 0.05 6.6 0.45 6.8 0.95 7.8 1.45 8.3 1.95 9.7 2.45 10 2.95 7.6 3.45 7.6 3.95 7.6 4.45 7.6 4.95 7.6"
filament_settings_id = ""
filament_soluble = 0
filament_toolchange_delay = 0
filament_type = PLA
filament_unload_time = 0
filament_unloading_speed = 90
filament_unloading_speed_start = 100
first_layer_bed_temperature = 65
first_layer_temperature = 200
inherits = 
max_fan_speed = 80
min_fan_speed = 100
min_print_speed = 10
slowdown_below_layer_time = 5
start_filament_gcode = "; Filament gcode\n"
temperature = 200

[printer:Geeetech A20M - E3D v6]
bed_shape = 0x0,255x0,255x255,0x255
before_layer_gcode = ;M117 Printing layer [layer_num]\n;LAYER:[layer_num]
between_objects_gcode = 
cooling_tube_length = 0
cooling_tube_retraction = 0
default_filament_profile = ""
default_print_profile = 
deretract_speed = 0
end_gcode = ; printing object ENDGCODE\n\nM117 Finished printing.\n\nM104 S0 ; turn off hotend\nM140 S0; turn off hotbed\n\nG28 X0  Y255 F9000; Feed the bed forward making removal easy.\nM84     ; disable motors
extra_loading_move = 0
extruder_colour = ""
extruder_offset = 0x0
gcode_flavor = marlin
high_current_on_filament_swap = 0
host_type = octoprint
inherits = 
layer_gcode = 
machine_max_acceleration_e = 800,5000
machine_max_acceleration_extruding = 800,600
machine_max_acceleration_retracting = 1000,600
machine_max_acceleration_x = 800,1000
machine_max_acceleration_y = 800,1000
machine_max_acceleration_z = 100,200
machine_max_feedrate_e = 25,120
machine_max_feedrate_x = 300,200
machine_max_feedrate_y = 300,200
machine_max_feedrate_z = 5,12
machine_max_jerk_e = 2.5,2.5
machine_max_jerk_x = 8,4
machine_max_jerk_y = 8,4
machine_max_jerk_z = 0.2,0.4
machine_min_extruding_rate = 0,0
machine_min_travel_rate = 0,0
max_layer_height = 0
max_print_height = 255
min_layer_height = 0.07
nozzle_diameter = 0.4
parking_pos_retraction = 0
print_host = 10.11.12.150
printer_model = 
printer_notes = 
printer_settings_id = 
printer_technology = FFF
printer_variant = 
printer_vendor = 
printhost_apikey = xxxx
printhost_cafile = 
remaining_times = 1
retract_before_travel = 2
retract_before_wipe = 0%
retract_layer_change = 0
retract_length = 2.1
retract_length_toolchange = 10
retract_lift = 0
retract_lift_above = 0
retract_lift_below = 0
retract_restart_extra = 0
retract_restart_extra_toolchange = 0
retract_speed = 40
serial_port = /dev/ttyUSB0
serial_speed = 250000
silent_mode = 0
single_extruder_multi_material = 1
start_gcode = M412 S0; Stacco sensore filamento\n\nM117 Homing\nG28 ; Home extruder\n\nM117 Auto bed levelling\nG29 L0        ; Load the mesh stored in slot 0 (from G29 S0)\nG29 J           ; Livellamento piano a 3 punti per correggere l'eventuale tilt\n \nM117 Printing...
toolchange_gcode = ;CAMBIO ESTRUSORE. Current=[current_extruder] Next=[next_extruder] Previous=[previous_extruder]
use_firmware_retraction = 0
use_relative_e_distances = 0
use_volumetric_e = 0
variable_layer_height = 0
wipe = 1
z_offset = 0

[presets]
print = Standard 0.20mm
sla_print = - default -
sla_material = - default -
printer = Geeetech A20M - E3D v6
filament = AmazonBasics PLA Viola
```

bubnikv, 2019-04-05:
The defaults for the extrusion width (if you let the extrusion widths zero)
may not be ideal. I recommend to derive your profiles from the Prusa3D
profiles.

In the future, we shall likely bake the Prusa3D tuned profiles into Slic3r
as the defaults.

guestisp, 2019-04-05:
I've calibrated the extrusion width multiple times.
I've tried with 0.4 (nozzle 0.4) and with default (0.45)
In both cases, a thin wall test has the perfect size but still exibits a sort of inconsistent extrusion, with layers more thicker than others and the overall surface is much more rough

I don't have this issue with S3D or Cura so i think there is a bad setting in Slic3r, but i don't know which one.

Any help would be apreciated.
I'll start a print with slic3r stable in less than 30 minutes, just to see if the beta has added some strangness

guestisp, 2019-04-05:
I did a gcode compare between Slic3rPE Stable, alpha7 and beta1.
In all cases i've exported the beta1 config bundle (currenlty, i'm using the beta1 as my primary slicer) to the stable and to the alpha, so the filament, printing and printer settings are the same.

alpha7 and beta1 export the same identical gcode (md5 hash is the same)
the stable has a different gcode.

I've compared the gcode between stable and beta1 and changes are mostly in Y positioning and, a lot of changes in extrusion amount, in example:

From the stable:
```
G1 X156.556 Y111.667 E5.20650
```
From the beta1:
```
G1 X156.563 Y111.676 E5.20698
```

guestisp, 2019-04-05:
I'm able to print a thin wall test properly, with absolutely no defects. Any other prints, like this rabbit: https://www.thingiverse.com/thing:220619 makes tons of lines, similiar to a wobble, but I don't have any wobble with this printer, tested by printing tens of wobble test with success.

I think there is something strange in calculating the positionings.

guestisp, 2019-04-08:
Any suggestion ?

guestisp, 2019-04-08:
By using `external_perimeters_first` seems to greatly improve the overall quality and no more badly aligned layers. After this benchy i'll print the same with `external_perimeters_first`  disabled (as per default) just to compare.

bubnikv, 2019-04-08:
You may also have an issue with over extrusion. Try to calibrate your
E-steps or set the extruder multiplier in Slic3r below 1.

guestisp, 2019-04-08:
E-steps and flow are calibrated. It's the first thing i did

The current benchy Is printing very very well, not visible artefacts (except some oozing due to the filament)

I'll reprint without external perimeter set just to confirm but there is a day and night difference to not consider this as bug.
Also, both S3D and Cura prints without external perimeters set and without any issues

guestisp, 2019-04-10:
I did another test, this should confirm that this is a slic3r issue/bug/whatever.

The following Benchy was printed with an Ender 3 Pro. A totally different printer than mine. Nothing shared between these two printer. Differerent hardware, different software (Marlin 1.1.6), different filament, different hotend. All different.

The only common thing is the gcode, i've used the same gcode as mine. So, something strange is placed in this gcode by Slic3r. I've attached the gcode here. Maybe there is a wrong setting ?
[3DBenchy_slic3r.zip](https://github.com/prusa3d/Slic3r/files/3065188/3DBenchy_slic3r.zip)
By using external perimeter first, print get a little bit better, but still not perfect. (and I can't use external perimeter as default as would make overhangs worse)

Look at this print and the look at mine, posted initially. They are almost the same, with the same bad layers in the same exact position.

[image]

guestisp, 2019-04-10:
If usefull, a Benchy gcode made with Cura is about 13MB (same layer height). The same made with Slic3r is about 2.3MB

guestisp, 2019-04-11:
Did another print. This time with my printer, same filament, same settings, but with Cura. The hull is smooth, no "banding" or similiar. On reddit there are a couple of post saying that Slic3r has trouble when slicing the Benchy, the posted pics are identical to mine, so it's a slic3r issue.

Would be possible to have a fix ? What's wrong with the benchy that is causing the layer banding ?

bubnikv, 2019-04-11:
I cannot do the homework for you. You bought a budget printer without a fine tuned Slic3r PE profile. There may be fine tuned Cura profiles available for your budget printer, so you may try to copy the parameters from Cura to Slic3r PE.

You seem to be printing with the "default" profile, which uses some auto magic derived from the nozzle diameter. I recommend to set these values similar to the Prusa3D profiles, depending on your nozzle diameter.

[image]

guestisp, 2019-04-11:
> I cannot do the homework for you.
> You bought a budget printer without a fine tuned Slic3r PE profile.
> There may be fine tuned Cura profiles available for your budget printer, so you may try to copy the parameters from Cura to Slic3r PE.

The same budget printer is used to print, without any optimized profile, gcode made with Cura and Simplify 3D. No issue at all.

The same gcode created from Slic3r was used to print to two different printers (an FDM and a Delta) with the same identical results in the same identical positions)

Budget printer or not, 3 different printers, totally unrelated, are printing the same gcode in the same bad way.

> You seem to be printing with the "default" profile, which uses some auto magic derived from the nozzle diameter.
> I recommend to set these values similar to the Prusa3D profiles, depending on your nozzle diameter.

I'm not printing with the default profile. It's more than 1 month that i'm trying to print a Benchy properly with Slic3r PE (and other prints too, with the same layer banding or bad extrusion)

Budget printer or not, the same **IDENTICAL** settings, checked one by one, ported from my Slic3r PE to Cura v4 and to Simplify 3D resulted in a correct print.
I've made, more or less, about 150 benchys using 2 spools of filament.
I've also did the same in the opposite way: copy a working Simplify3D settings to Slic3r PE: same bad benchy.

So, no, you don't have to do homeworks for me but please take this issue seriously.

ManuelMcLure, 2019-04-11:
On the blue Benchy, the very noticeable "waterline" is probably cooling related. In your cooling settings, make sure "Keep fan always on" is checked, and for PLA, set fan min speed to at least 75% (or even test 100%). Cura generally leaves the fan on 100% for all layers except the first so that's a big difference. I noticed the same "waterline" the first time I tried printing a Benchy with Slic3r after having exclusively used Cura.

guestisp, 2019-04-11:
I'm using fan at 100%. But it's not the waterline the issue. Look upper, there are a huge "banding".

bubnikv, 2019-04-11:
install the prusa3d mk2 profile and derive from it. change the start gcode
and increase retraction and you should be good to go.

guestisp, 2019-04-11:
I'll try but i don't think it's a simple settings issue

bubnikv, 2019-04-11:
you seem to be the only one having such a severe issue with the current
slic3r pe beta. My only logical conclusion is that your settings is
suboptimal.

guestisp, 2019-04-11:
It's not the beta. I have the same with all slic3r prusa, stable, alpha and beta. 
Slic3r non prusa is a little better but still have inconsistent extrusion issues, especially (but not limited to) with the benchy

guestisp, 2019-04-11:
Here Is another print where you can see a huge seam and the inconsistent extrusion.
It's not a wobble, because is not uniform and change at every print

[image]

guestisp, 2019-04-11:
Another print but this time with Prusa MK2 settings, as per your suggestion
Mostly identical to the previous one with my custom settings, except for the seam (probably i've forgot to enable the vase mode)

[image]

guestisp, 2019-04-19:
Ok, i've found the exact cause of these banding: the seam
Every time the seam is placed on a corner, a huge banding is printed. I've tried to print the same object with seam forced on the rear and all surfaces (except the rear one with the seam) are totally flat.

So, the question is: why with Slic3r I have a huge seam (very visible if placed on the rear) or a huge banding when the seam is on the corner ?

The retraction is correct, I don't see any oozing or stringing during the retraction tests.

skellatore, 2019-04-30:
Stepper Drivers?
Acceleration?
Jerk?
Feed rate?
All of these come into play. 

It took months (no exaggeration) of getting my Brand new CR10S S5 to print properly. The belts sucked, the extruder sucked, the bed was a heavy dead weight that had more waves than the ocean, the whole bench shook and vibrated and the gantry required stiffening, the Z-Axis' dropped on one side when the stepper motors were off.... the list goes on.

If your settings are so awesome from Cura/Simplify3D - why haven't you copied the print speed/acceleration/jerk/feed rates/Extrusion ratio/start and end GCODE/extrusion width/extrusion height and other settings into Slic3r PE? Not doing this and expecting the same results as the *OTHER* software is absurd.

guestisp, 2019-04-30:
I've tried the same settings....

Jbd-aix, 2019-07-03:
Hey just wanted to let you know that i have the exact same issued, but with the new prusaslicer and Cura 4.0/4.1. I have tried marlin 1.1.9 and 2.0.x (new 32bit Board), but nothing changed. 
Calibrating linear advance helped a bit though.
Have you found a solution in the meantime?

guestisp, 2019-07-04:
No solution other than using a different slicer for important prints. I use Slic3r only for drafts

ajayjohn, 2019-09-09:
I too have practically identical problems with my benchys when using PrusaSlicer 2.1 RC. Just as reported here, the issue is non-existent when using the Cura. In fact, the banding/waterlines even appear at the exact same positions as the one shown in the Ender 3 Pro print. 

I wish the issue wasn't simply brushed away attributing it to configuration error/printer issues. I understand that most errors may arise from that, but guestisp here seems to have been quite systematic in reporting the error and testing the issue under various conditions.

guestisp, 2019-09-09:
As i'm not using an "original Prusa" i'm not taken in consideration, every time i open and issue, it's being ignored or delayed for years

So, the solution is to use a different slicer, very simple. Cura works great and even my regural licensed s3d

liftbag, 2020-01-26:
Not only I, but a lot of people in an italian PrusaSlicer telegram group have this issue.
I am as comfortable with PrusaSlicer as I am very uncomfortable with Cura. But I've been fighting with this issue since the time of Slicer PE. I also notice a certain disinterest for those who report issues without owning a Prusa 3d printer.
But I totally agree with @guestisp: prusaslicer creates artifacts on the walls that other slicers don't create. Even if this was caused by incorrect settings, it is absurd that it is so complicated to obtain results comparable to other slicers playing with the parameters, results that in years of almost exclusive use of PS, I still have not obtained.
In the hope that this request will be really taken into consideration, I will also have to overcome the learning curve of slicers which I consider substantially inferior to PrusaSlicer.

bigsubgit, 2020-01-27:
I'm having the same problems I suppose. Just a test with the same parameters, sliced with S3D, doesn't reply the problems.
[image]
[image]

On the left sliced with ps.

banzaito, 2020-01-28:
Layer alignment is a big problem that forces me not to be able to use PS. With S3D, which I have licensed, I have no problems but also with others like Kisslicer and Cura the layers remain very well aligned. You are doing an excellent job but this is a serious problem ...... I hope you can fix it soon.

3displit, 2020-01-28:
Reporting same issue on Tronxy X5SA.

With cura it's perfect, with Slic3r PE same as others on images.

lukasmatena, 2020-01-29:
@liftbag @bigsubgit @banzaito @3displit and others

Guys, you provided virtually no useful information, apart from "PrusaSlicer sucks". Please understand, that many people claim that PrusaSlicer is actually same or better that competition slicers (quality-wise), so it is hard to imagine this would be a flaw in the software itself. And because most people here have printers, that we don't supply profiles for, it strongly suggests a profile-related problem. We have dozens of printers running 24/7 and if the quality shown e.g. on the first benchy here was inherent to PS, we would have noticed.

If anyone sees this on an MK3 with our stock profile or a profile that is close to it, provide a 3MF (the simpler, the better), gcode and PrusaSlicer version. Then we can try to reproduce it on another printer. But I cannot debug an issue based on "it prints bad with a printer you never heard of, with a profile you didn't see and I won't tell you where it came from and what version of PS I used - just fix it". That is simply impossible.

liftbag, 2020-01-29:
> @liftbag @bigsubgit @banzaito @3displit and others
>
> Guys, you provided virtually no useful information, apart from "PrusaSlicer sucks". Please understand, that many people claim that PrusaSlicer is [...]
Sorry, but I thought that the amount of information provided by @guestisp was largely sufficient. We who complain about the same differences in results between PrusaSlicer and the other slicers, have intervened in the comments to bring up the attention on the issue. Whether it's a slicing algorithm problem or a settings problem, it still remains a problem. If this does not happen with profiles for prusa printers on original prusa printers, then it means that PrusaSlicer is highly optimized for prusa printers, as is perhaps right. In this case I take note of it and will act as it suits me.
The problem mentioned in this issue is known to me from the first uses of SlicerPE, but I always thought it was a problem of my hardware. It is only recently, driven by the complaints expressed in the Italian PrusaSlicer Telegram group (founded by me) that I investigated better discovering the issue is not hardware, but software.
And I want to say that I don't identify with those who say that prusaslicer sucks at all. After purchasing the license for S3D, I discovered SlicerPE, when there were still very large gaps with the supports management. Yet I liked the software so much that I completely abandoned S3D. Still today I consider it much superior to S3D and very ergonomic compared to Cura.
For my part, I will invest some more time to carry out tests by providing all the useful documentation, without prejudice whatsoever.

3displit, 2020-01-29:
Prusa is awesome and I posted here because I love it and want to continue to use it.

I did all the troubleshootings. 
Finally reinstalled prusa slicer and deleted profiles before.
Sliced same object as before and...it worked. Looks like it is fixed for me.

But not sure what was wrong. Because I used default profiles before and now.

liftbag, 2020-01-29:
> Prusa is awesome and I posted here because I love it and want to continue to use it.
>
> I did all the troubleshootings.
> Finally reinstalled prusa slicer and deleted profiles before.
> Sliced same object [...]
Thanks for your report. I will try as soon as possible what you suggest.

banzaito, 2020-01-29:
> @liftbag @bigsubgit @banzaito @3displit and others
>
> Guys, you provided virtually no useful information, apart from "PrusaSlicer sucks". Please understand, that many people claim that PrusaSlicer is [...]

https://github.com/prusa3d/PrusaSlicer/issues/2070#issuecomment-579645384

> @liftbag @bigsubgit @banzaito @3displit and others
>
> Guys, you provided virtually no useful information, apart from "PrusaSlicer sucks". Please understand, that many people claim that PrusaSlicer is [...]

I am sorry I did not provide enough data to help you solve the problem but I also thought you had enough data given the previous reports also I would like to underline that I congratulated you for the work done so far by pointing out that in any case this problem makes Prusa Slicer unusable ..... never said that Prusa Slicer sucks. I will also try the 3displit solution although it does not seem to be a "solution" ..... it would obviously be better to be able to customize the profiles for your printers, in my case with slicers competing with these three different kinematics I get excellent performance.

My printers:
CoreXY self build with Duet2Wifi - RepRap Firmware
Delta kossel self build with Duet2Wifi - RepRap Firmware
Cartesian Tatara frame with Duet2Wifi - RepRap Firmware

liftbag, 2020-01-30:
As promised, I made some test to identify possible causes of what looks like a misalignment due to the slicer, and I have come to a conclusion. It seems that the automatic ventilation, together with my very powerful fanduct, causing differences in the aesthetic of the outer perimeters. By setting a fixed ventilation, the defects disappear. This is enough to be satisfied and to be able to continue using PrusaSlicer, as well as to continue to support feature requests on interesting functions, such as ironing and fuzzy skin.

This was my first test, with ventilation set to auto. Similar results with S3D, set by copying the PS settings.
[image]

This is the result by setting manual fixed ventilation.
[image]

Same settings on S3D, where you can see the hateful seams.
[image]

guestisp, 2020-02-01:
The issue is mostly caused by speed changes, by using slow perimeters or external perimeters much slower than other (as default, there is at least a 50% slowdown)

The same happens by using auto ventilation, that slow down the print as well .

Disabling auto ventilation and setting slow perimeters as external perimeters (5% less than internal) mostly fix the issue

Tried the same with multiple printers, multiple firmware, multiple filaments , multiple extruders, multiple hotend (cinese and original), multiple board (lerdge, geeetech, bigtrretech), both 8 and 32bit. All the same

The issue is prusaslicer related, is not sn hardware issue

bigsubgit, 2020-02-01:
> The issue is mostly caused by speed changes, by using slow perimeters or external perimeters much slower than other (as default, there is at least a 50% slowdown)
>
> The same happens by using auto [...]

Hi, I never use auto ventilation and I usually use the same speed for all kinds of perimeter (25mm/s), but I have the problem. 
I also tested with default mk3s profile (I've an ender 3) but again, bad surface exeternal. 
I don't know anymore...

guestisp, 2020-02-02:
Yes, the issue is still present, speed changes only make it even worse

kxu-ffd, 2020-02-11:
Reporting the same issue with original prusa mk3s and PrusaSlicer 2.1.1 and when switched to Cura 4.3 and sliced the same part, all the layer misalignments are gone. Here are the comparisons.
**PrusaSlicer 2.1.1**
[front] https://drive.google.com/file/d/1Jw2eigbw1RUIVI8YlQSn2B1sTAJytpqw/view?usp=sharing
[back] https://drive.google.com/file/d/1zbFvyYI3Kvh2P2pEezOhDIcgVxPdCwkH/view?usp=sharing
**Cura 4.3**
[front] https://drive.google.com/file/d/1cNrRaWZBtfblCRm7p10S7MYeF5YNdIQZ/view?usp=sharing
[back] https://drive.google.com/file/d/1jUVciRY-3dHtHdlBI7gVP-0J3ajpqu9-/view?usp=sharing
**Files**
[stl] https://drive.google.com/open?id=1Bsmrc0HMrUJmojl1zCmHOTJ9brbAx03D
[3mf] https://drive.google.com/open?id=1uA4yq0xkyWVTHcHs6O_oq0BT7z2XqTqo

Shamoon78, 2020-04-16:
Extruder Jerk 2.5 is low try to set it to 5 or more, and try to be sure your Acceleration max is 3000,3000,100,5000
printing acceleration 800
jerk 10,10,0.4,5
I used default cr10 profiles in cura I got same artifacts with sever banding but when switching to prusa profiles for cura and adjust the firmware parameters as above I got perfect print only very tiny stringing in the windows of the benchy but fixed by increase the retraction for prusa profiles 0.8 to 2 mm while cr10 profile already use 6.5 mm 
my printer is core xy self designed and made with direct drive extruder heavily modified mk2 prusa design extruder to fit micro Swiss hot end of cr10 pro instead of e3d v6 and slightly larger stepper motor gear to create a slight bending in filament ,which for me out perform the mk3 prusa extruder design with dual gears that always gave me inconsistent extruding. and drive me crazy for couple months of testing.

daleckystepan, 2020-09-13:
Hi guys,
I have experienced very similar/same issue.

What helped me a bit:
* decrease printing temperature about 5 degrees
* tune extrusion multiplier or steps per mm on extruder  - usually decrease it

What nearly resolved issue:
* print all perimeters over the height with the same print speed
* print all external perimeters over the height with the same speed

Variation in speed creates layer misalignments. See attached screenshot.
[image]
[image]

boelle, 2020-10-01:
could you attach the actual setting page? have the same issue

bubnikv, 2020-11-30:
This issue is stalled. We believe most of the issues mentioned are not PrusaSlicer issues.
There is also a write-up on some of the issues you are seeing and why they are happening https://www.help.prusa3d.com/en/article/the-benchy-hull-line_124745/

Closing

closed 2020-11-30 by bubnikv

Joaquin-hue, 2021-02-03:
Pudieron Solucionarlo?

Joaquin-hue, 2021-02-03:
Hace 4 Meses que estoy teniendo este mismo error