prusa3d/PrusaSlicer #2071: simple small print - crash on slice

akust, 2019-04-04:
### Version
1.42.0 beta

### Operating system type + version
OSX 10.12.6 OS Sierra

### 3D printer brand / version + firmware version (if known)
stock

### Behavior
* _Describe the problem_
stl comes from OpenSCAD, extreme simple cube
any setting in Slic, trigger Rende.. and it crashed
I checked the opposite (large complicated detailed 2 color print) Render - works
again loaded the simple test stl, render - immediate crash

* _Steps needed to reproduce the problem_
try to render attached file/content in zip
ok, it's so simple:
- open OpenScad, enter:
linear_extrude(height=0.4, center=true) {square(size=[40, 40], center=true);}
..and render it as stl.
- open Slic3r, everything to untouched default prusa original original settings
- render
--> 100% crashes here

* _Expected Results_
1-layer gcode rendering

* _Actual Results_

well.. this:

"
rocess:               Slic3r [20016]
Path:                  /Applications/Original Prusa Drivers/Slic3r 142beta.app/Contents/MacOS/Slic3r
Identifier:            com.prusa3d.slic3r/
Version:               Slic3r Slic3rPE-1.42.0-beta+full-201903171402 (Slic3rPE-1.42.0-beta+full-201903171402)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Slic3r [20016]
User ID:               502

Date/Time:             2019-04-05 00:33:24.185 +0200
OS Version:            Mac OS X 10.12.6 (16G1710)
Report Version:        12
Anonymous UUID:        CBA1276C-1110-671D-B589-EE0037C6866F

Sleep/Wake UUID:       CE9D7526-825C-44EC-A25A-99E7B4F06746

Time Awake Since Boot: 100000 seconds
Time Since Wake:       18000 seconds

System Integrity Protection: enabled

Crashed Thread:        6

Exception Type:        EXC_ARITHMETIC (SIGFPE)
Exception Codes:       EXC_I386_DIV (divide by zero)
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Floating point exception: 8
Termination Reason:    Namespace SIGNAL, Code 0x8
Terminating Process:   exc handler [0]

Thread 0:: Dispatch queue: com.apple.main-thread
0   com.apple.CoreGraphics        	0x00007fff81f012f9 CGGStateGetStyle + 1
1   com.apple.CoreGraphics        	0x00007fff81ee7244 ripc_GetRenderingState + 313
2   com.apple.CoreGraphics        	0x00007fff81ee9d64 ripc_DrawImage + 213
3   com.apple.CoreGraphics        	0x00007fff81f6226a CGContextDelegateDrawImage + 48
[... 437 lines cut ...]
shared memory                     16.4M       13 
===========                     =======  ======= 
TOTAL                            949.1M      662 

Model: MacBookPro12,1, BootROM 180.0.0.0.0, 2 processors, Intel Core i5, 2.7 GHz, 16 GB, SMC 2.28f7
Graphics: Intel Iris Graphics 6100, Intel Iris Graphics 6100, Built-In
Memory Module: BANK 0/DIMM0, 8 GB, DDR3, 1867 MHz, 0x80AD, 0x483943434E4E4E434C544D4C41522D4E5544
Memory Module: BANK 1/DIMM0, 8 GB, DDR3, 1867 MHz, 0x80AD, 0x483943434E4E4E434C544D4C41522D4E5544
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x133), Broadcom BCM43xx 1.0 (7.21.171.133.1a2)
Bluetooth: Version 5.0.5f3, 3 services, 27 devices, 1 incoming serial ports
Network Service: Wi-Fi, AirPort, en0
Serial ATA Device: APPLE SSD SM0256F, 251 GB
USB Device: USB 3.0 Bus
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 27.1
"

  * _Screenshots from __*Slic3r*__ preview are preferred_
sorry, opening preview triggers the same as Render (see *_actual Result_*)

_Is this a new feature request?_
nope, but a crash

#### Project File (.3MF) where problem occurs
[testprint_crashes slic3r_files.zip](https://github.com/prusa3d/Slic3r/files/3045719/testprint_crashes.slic3r_files.zip)

BerndJM, 2019-04-04:
Can confirm the crash with the -beta ... but it looks like all is fine with the -beta**1**

bubnikv, 2019-04-05:
thanks. this issue was fixed in beta1 indeed. closing.

closed 2019-04-05 by bubnikv

akust, 2019-04-05:
oh, then  "beta" is not "beta1"
maybe you could name it more differently, like beta0, then beta1 or beta, then beta2..
would have saved me countless crashes and triels and 30min of writing bugreport.. thx