prusa3d/PrusaSlicer #2072: Model repair creates top-/bottom-layer when it shouldn't

Rick-T, 2019-04-05:
### Version
Commit [ccee04cd689017ba4880e6cd74d8c170ac40e7af](https://github.com/prusa3d/Slic3r/commit/ccee04cd689017ba4880e6cd74d8c170ac40e7af)

I'm using the AppImage that was linked in [this issue](https://github.com/prusa3d/Slic3r/issues/2069)

The problem also occurs in the AppImage from the releases 1.42.0-beta1 **and** 1.41.3.

### Operating system type + version
**Operating System**: Unix
**System Architecture**: 64 bit
**System Version**: Linux 4.20.17-1-MANJARO x86_64

### 3D printer brand / version + firmware version (if known)
Prusa MK3 MMU2

### Behavior
Slic3r fills a whole in an object with a top- or bottom-layer, when it should really be empty. I noticed the error on the Top_Band.stl file from [this model on thingiverse](https://www.thingiverse.com/thing:2379204/files).

When I import the object it's basically a "bridge" with two holes. When I slice it, everything looks fine as well. But when I turn the object on it's side, one of the holes gets filled. Please take a look at the screenshots for details.

The object itself:
[image]

When I just slice the object, it looks fine. Two holes, everything's nice:
[image]

When placed on the side, suddenly one of the holes is filled:
[image]

When placed on the other side, the same hole is filled again:
[image]

#### Project File (.3MF) where problem occurs
[Top_Band.3mf.gz](https://github.com/prusa3d/Slic3r/files/3048416/Top_Band.3mf.gz)

Rick-T, 2019-04-05:
While writing this issue I noticed that there is already a [similar issue](https://github.com/prusa3d/Slic3r/issues/1839) about model fixing.

In that issue @bubnikv said `We keep a database of broken models, and every time we change the heuristics we re-test over our database. (...) I will keep this issue open, so one day we can return to it and possibly improve the heuristics for fixing slices over broken parts of the model.`

I did not want to waste all my writing, so I posted this issue anyways. To the devs: Are you interested in this kind of report? How do you want to be informed about these kinds of errors? In the future I don't want to annoy you by opening an issue for every model-fixing error that I find.

bubnikv, 2019-04-05:
You may open a new issue, or you may just piggyback onto another issue, if
you believe it is the same one. For such an issue, we need to collect data
and do regression tests after an apparent fix. To slice a broken model is a
guesswork, and a fix for one set of model errors may break slicing for
another broken models.

bubnikv, 2019-04-13:
Slic3r indicates errors in the model. If the model is fixed with the netfabb service (just click on the exclamation mark on the side bar with a right mouse button on Windows 10), the file slices correctly.

I am leaving this issue open, so we have a training material for improving the model fixing.

edgarjs, 2020-10-09:
I'm having a similar issue, and also am curious of how should we report when the model is licensed? I would guess it'd need to be via email? I'm not sure, but here's a screenshot of the pre-sliced model vs the sliced one with the default settings for MMU single.

[image]

[image]

closed 2026-09-29 by prusaslicer-bot[bot]