prusa3d/PrusaSlicer #2073: Variable layer height feature breaks after rotating part

rohrbrecher, 2019-04-05:
### Version
1.42.0 beta1 win64

### Operating system type + version
Windows 10 64bit

### Behavior
Variable layer height only works as long as the part is not rotated around X/Y or via "Place on face".

Positive test:
Step 1: Import stl
Step 2: Modify layer height in "Layers editing"
Step 3: Observe that layer height changed in sliced view.

[image]

Negative test:
Step 1: Rotate part around X/Y or place on some other face.
Step 2: Repeat "Layers editing" and change the height of some part.
Step 3: Observe that layer height has not changed in sliced view.

[image]

I observed this on various stl files so I guess it's not a problem of the file.

bubnikv, 2019-04-13:
Thanks, I can reproduce on many objects, for example on this tiny cone.

[cone.zip](https://github.com/prusa3d/Slic3r/files/3075935/cone.zip)

bubnikv, 2019-04-13:
There was an approximate bounding box used at the GUI, while a snug
bounding box was used at the back end, causing invalidation
of the variable layer height editing profile on rotated objects.

Fixed with 1e455bc0654c4764e7231b9d41a919be065e7dcd
Closing.

closed 2019-04-13 by bubnikv