prusa3d/PrusaSlicer #2074: "Plater" is misspelled

zanderk, 2019-04-06:
### Version
_slic3rPE-1.42.0-beta1+linux64-full-201903231838.AppImage

### Operating system type + version
Linux 

### 3D printer brand / version + firmware version (if known)
Prusa

### Behavior
In English, "Plater" is spelled "Platter" (missing one "t").

_Is this a new feature request?_

BerndJM, 2019-04-06:
#1226

cosmith71, 2019-04-07:
It's common in English to add -er to a verb to make a noun.  I always assumed that you're arranging things on the build plate so you are plating them.  Therefore the thing that does that would be the plater.

StephenRC, 2019-04-07:
On Sun, 07 Apr 2019 04:31:02 -0700, Colin Smith
<notifications@github.com> had a flock of green cheek conures squawk
out:

>It's common in English to add -er to a verb to make a noun.  I always assumed that you're arranging things on the build plate so you are plating them.  Therefore the thing that does that would be the plater.

plat·ing
/'pladiNG/
noun
1. a thin coating of gold, silver, or other metal.
2. an outer covering of broad, flattish sections, typically of metal.
"the tractors carried steel plating for protection"
----------------------------------------------------------------------------------------------
plater
1 : one that plates
2a : a horse that runs chiefly in plate races
b : a racehorse that competes in the lowest grade of racesplat

----------------------------------------------------------------------------------------
/plat/
NORTH AMERICAN
verb
gerund or present participle: platting

plan out or make a map of (an area of land, especially a proposed site
for construction).
"he bought back the site, platted it, and named it after his realtor"
---------------------------------------------------------------------------------------------
plat·ter
/'plad?r/
noun
noun: platter; plural noun: platters
1.a large flat dish or plate, typically oval or circular in shape,
used for serving food.
synonyms:	plate, dish, serving plate, serving dish, salver,
tray; More
a quantity of food served on a platter.
"huge platters of cold cuts"
a meal or selection of food placed on a platter, especially one served
in a restaurant.
"I'll have the seafood platter"
2.the rotating metal disk forming the turntable of a record player.
INFORMALDATED
a phonograph record.
COMPUTING
a rigid rotating disk on which data is stored in a disk drive; a hard
disk (considered as a physical object).

-- 

Stephen

. . . . File not found. Should I fake it? Y/N

---
This email has been checked for viruses by AVG.
https://www.avg.com

cosmith71, 2019-04-07:
From:  https://www.merriam-webster.com/dictionary/plate

plate noun
\ ˈplāt  \
Definition of plate (Entry 1 of 2)
**1a : a smooth flat thin piece of material**
b(1) : forged, rolled, or cast metal in sheets usually thicker than ¹/₄ inch (6 millimeters)
(2) : a very thin layer of metal deposited on a surface of base metal by plating
c : one of the broad metal pieces used in armor
also : armor of such plates
d(1) : a thin relatively flat anatomical part (such as a lamina of bone) of an animal body
especially : SCUTE
(2) : the thin under portion of the forequarter of beef
especially : the fatty back part
— see BEEF ILLUSTRATION
e : HOME PLATE
f : any of the large movable segments into which the earth's lithosphere is divided according to the theory of plate tectonics
2 [ Middle English; partly from Anglo-French plate plate, bullion; partly from Old Spanish plata silver, from Vulgar Latin *platta metal plate, from feminine of plattus flat ]
a obsolete : a silver coin
b : precious metal
especially : silver bullion
3 [ Middle English, from Anglo-French plat, plate dish, plate, from plat flat ]
a : domestic hollowware made of or plated with gold, silver, or base metals
b : a shallow usually circular vessel from which food is eaten or served
c(1) : a quantity to fill a plate : PLATEFUL
(2) : a main course served on a plate
(3) : food and service supplied to one person
a dinner at $10 a plate
d(1) : a prize given to the winner in a contest
(2) British : a horse race in which the contestants compete for a prize of fixed value rather than stakes
e : a dish or pouch passed during collections
f : a flat glass or plastic dish used chiefly for culturing microorganisms
**4a : a prepared surface from which printing is done**
b : a sheet of material (such as glass or plastic) coated with a light-sensitive photographic emulsion
c : a metallic grid with its interstices filled with active material that forms one of the structural units of a battery
d : LICENSE PLATE
5 : a horizontal structural member that provides bearing and anchorage especially for the trusses of a roof or the rafters
6 : the part of a denture that fits to the mouth
broadly : DENTURE
7 : a full-page illustration often on different paper from the text pages
8 : a schedule of matters to deal with
have a lot on my plate now
on a plate
: without having been earned : as a gift
goals were handed to them on a plate
plate verb
plated; plating
Definition of plate (Entry 2 of 2)
transitive verb

1 : to cover or equip with plate: such as
a : to provide with armor plate
b : to cover with an adherent layer mechanically, chemically, or electrically
also : to deposit (something, such as a layer) on a surface
2 : to make a printing surface from or for
3 : to fix or secure with a plate
4 : to cause (a runner) to score in baseball
5 : to arrange (food) on a plate or dish

zanderk, 2019-04-07:
cosmith71:
> It's common in English to add -er to a verb to make a noun. I always assumed that you're arranging things on the build plate so you are plating them. Therefore the thing that does that would be the plater.

I'm being pedantic here but think how you would pronounce that: plater --> plate * er, one who plates.
platter --> plat * ter, something you put the turkey on.

First is a verb, second is a noun (as you noted). The thing we print on is a noun, not a verb. "The nozzle prints on the platter" is different than "The nozzle prints on the plater". Read those out loud.

cosmith71, 2019-04-07:
> cosmith71:
>
>
> I'm being pedantic here but think how you would pronounce that: plater --> plate * er, one who plates.
> platter --> plat * ter, something you put the turkey on.
>
> First is a verb, second [...]

I'd say "The nozzle prints on the plate".  The activity of arranging objects on the plate would be plating, and the part of the software that you use for plating would be the plater.

You're describing the build surface (platter is technically fine in that context), I'm describing the software in which objects are arranged, i.e., plated.  Which sounds much better than platting on the platter.  ;)

zanderk, 2019-04-07:
Good point, "plate" would be a fine (and probably better) term for that
flat thing parts grow from (up or down).

cosmith71, 2019-04-07:
I'm sure the non-native English speakers are having fun watching us argue over a language where there are no rules and everything is made up randomly anyway.  😄

guestisp, 2019-04-07:
> I'm sure the non-native English speakers are having fun watching us argue over a language where there are no rules and everything is made up randomly anyway

Yes, that's funny :D :D :D

BerndJM, 2019-04-07:
If the software has bugs or issues I think it's not really relevant if this bugs/issues occur on a plater or a platter ...  :smile: :smile: :smile:

BirkBinnard, 2019-04-08:
Plater is correct because the object on which the print is made is called the build plate. Consequently the function of arranging what is to be printed is plating. For comparison, the function of arranging houses on subdivision lots, or plats, is called platting. 

Therefor the software that does the arranging of objects to be printed should be called the plater. 

And if I may be allowed to beat a dead horse, it is true that a turkey is served on a platter. But the carver, who slices the bird and places the slices on dinner plates to be served, is also called the plater.

bubnikv, 2019-04-08:
Hey, Birk, you are Dutch, right? I admire your English skills.

This discussion is hilarious. It shows that Slic3r must be in a great
shape, if this "issue" seem to have such a high priority to some.

BirkBinnard, 2019-04-08:
Actually I am American - my ancestors came to the US in the early 1800's and ended up being provisioners for the Gold Rush miners in the state of Idaho in the 1850's. But I have been to the Netherlands once - a fascinating place for sure. And although I majored in math & computer science my best subject always was English. The small book titled "Elements of Style", a true classic, tells how to correctly form proper English sentences.

bubnikv, 2019-04-13:
Thank you guys for bringing some fun into our dry software development work. I think we had enough though.

I have created a FAQ page, so those interested can find the discussion and continue in the fruitful debate.
https://github.com/prusa3d/Slic3r/wiki/Frequently-Asked-Qeustions-(FAQ)

closed 2019-04-13 by bubnikv

Kaelum, 2024-09-13:
`Plater` is a verb, referring to a person plating a table, or placing food on a plate.

`Platter` is a noun, referring to built plate that you are building on, thus `Platter` is the correct word.