prusa3d/PrusaSlicer #2075: Can't resize platter window

zanderk, 2019-04-06:
### Version
slic3rPE-1.42.0-beta1+linux64-full-201903231838.AppImage 

### Operating system type + version
Linux

### 3D printer brand / version + firmware version (if known)
Prusa

### Behavior
Can't resize platter window.
This really sucks when you have a smallish or narrow display.

_Is this a new feature request?_

#### Project File (.3MF) where problem occurs
_Upload a Slic3r PE Project File (.3MF) (`Plater -> Export plate as 3MF` for Slic3r PE 1.41.2 and older, `File -> Save` / `Save Project` for Slic3r PE 1.42.0-alpha and newer)_

bubnikv, 2019-04-13:
> Can't resize platter window.

What do you mean with that? What cannot you resize? The main window? Is the side panel too wide? A screenshot would help.

zanderk, 2019-04-13:
[image: platterScreen1.jpg]
[image: platterScreen2.jpg]

bubnikv, 2019-04-13:
sorry, the screenshots are missing

zanderk, 2019-04-13:
Do you have a place I can upload them to? I'll try a different attachment
method.

eid0, 2019-07-29:
Me too, in Linux Plasma in plater tab, the panes can't be resized.
If you have configured somewhat big font size in system, then the 3d window is only a third of the screen (specially in layer mode).
It would be nice if the 3d window could be resized to better view the part. thx.

bubnikv, 2019-07-30:
> If you have configured somewhat big font size in system,

A screenshot would help, and you should likely make the system font smaller.

SolidRick, 2020-03-23:
If you have a 4:3 monitor, the sidepanel is too wide and should be resizeable, see my monitor
Also the Z-section slider takes too much width, could be helpful to move the Z-height and section Nr. somewhere else
[image]

foreachthing, 2020-03-23:
> Me too, in Linux Plasma in plater tab, the panes can't be resized.
> It would be nice if the 3d window could be resized to better view the part. thx.

Same for Windows.

JdeV987, 2020-06-06:
I have the same issue. Not everybody have a huge wide screen. Is it possible to add this feature in the future?

bubnikv, 2020-10-17:
The side panel could newly be collapsed in PrusaSlicer 2.3.0-alpha1. Closing.

closed 2020-10-17 by bubnikv