prusa3d/PrusaSlicer #2076: [Feature Request] Seam visualisation in Slic3r

rmoravcik, 2019-04-06:
Hi all,

would it be possible to add a visualisation of seam to the preview tab?

Thank youi.

Spacemarine2018, 2019-04-06:
In the preview, you can enable so see retractions and unretractions. Does that solve your problem?

rmoravcik, 2019-04-06:
@Spacemarine2018 I know about that setting in preview, but if you have retractions not only on layer change, than it's not so obious where the seam will be present:

[image]

guestisp, 2019-05-03:
+1 for this. I'm fighiting some issue with seam and retractions and this would be a great addition.

zalexzperez, 2020-03-16:
Yep, I second this

rtyr, 2021-05-03:
Implemented on master. Closing.

closed 2021-05-03 by rtyr

Hologos, 2021-09-19:
Has this been released yet? I can't seem to find in in latest release. Thanks

rtyr, 2021-09-19:
https://github.com/prusa3d/PrusaSlicer/releases/tag/version_2.4.0-alpha1