prusa3d/PrusaSlicer #2077: Windows build instructions broken

lf-, 2019-04-06:
### Version
_Version of Slic3r Prusa Edition used goes here_
master (2487bb8794e550ae1174e3cdb33fb7138d1dec04)

_Use `About->About Slic3r` for release versions_

_For -dev versions, use `git describe --tag` or get the hash value for the version you downloaded or `git rev-parse HEAD`_

### Operating system type + version
_What OS are you using, and state any version #s_
Windows 10

_In case of 3D rendering issues, please attach the content of menu Help -> System Info dialog_

### Behavior
* _Describe the problem_
When building on Windows, wxwidgets is not found by cmake

* _Steps needed to reproduce the problem_

Install a new Windows 10 (I recommend modern.ie VMs), install VS Community 2017 (this probably doesn't matter as to the version), and the latest cmake from the linked site in the build docs.

Download the compile stuff and place it somewhere.

Run `cmake .. -G "Visual Studio 15 2017" -DCMAKE_PREFIX_PATH="C:\whereyouputthestuff\usr\local"`

Logs:

[CMakeError.log](https://github.com/prusa3d/Slic3r/files/3050197/CMakeError.log)
[CMakeOutput.log](https://github.com/prusa3d/Slic3r/files/3050198/CMakeOutput.log)

```
PS C:\Users\lf\dev\slic3r\build> cmake .. -G "Visual Studio 15 2017" -DCMAKE_PREFIX_PATH="C:\Users\lf\dev\slic3r-devstuff\usr\local\"
-- Selecting Windows SDK version 10.0.17134.0 to target Windows 10.0.18356.
-- SLIC3R_ASAN: OFF
-- SLIC3R_BUILD_SANDBOXES: OFF
-- SLIC3R_BUILD_TESTS: OFF
-- SLIC3R_FHS: OFF
-- SLIC3R_GTK: 2
-- SLIC3R_GUI: ON
-- SLIC3R_MSVC_COMPILE_PARALLEL: ON
-- SLIC3R_MSVC_PDB: ON
-- SLIC3R_PCH: ON
-- SLIC3R_PERL_XS: OFF
-- SLIC3R_PROFILE: OFF
-- SLIC3R_STATIC: ON
-- SLIC3R_SYNTAXONLY: OFF
-- SLIC3R_WX_STABLE: OFF
-- SLIC3R_STATIC: ON
-- CMAKE_PREFIX_PATH: C:\Users\lf\dev\slic3r-devstuff\usr\local\ (from cache or command line)
Building without Win10 Netfabb STL fixing service support
CMake Error at C:/Program Files/CMake/share/cmake-3.14/Modules/FindBoost.cmake:1373 (file):
  file STRINGS file
  "C:/Users/lf/Dev/slic3r-devstuff/destdir-64-dev/usr/local/include/boost-1_63/boost/version.hpp"
  cannot be read.
Call Stack (most recent call first):
  CMakeLists.txt:229 (find_package)

CMake Error at C:/Program Files/CMake/share/cmake-3.14/Modules/FindBoost.cmake:2132 (message):
  Unable to find the requested Boost libraries.

  Boost version: 0.0.0

  Boost include path:
  C:/Users/lf/Dev/slic3r-devstuff/destdir-64-dev/usr/local/include/boost-1_63

  Could not find the following static Boost libraries:

          boost_system
          boost_filesystem
          boost_thread
          boost_log
          boost_locale
          boost_regex

  No Boost libraries were found.  You may need to set BOOST_LIBRARYDIR to the
  directory containing Boost libraries or BOOST_ROOT to the location of
  Boost.
Call Stack (most recent call first):
  CMakeLists.txt:229 (find_package)

CMake Error at cmake/modules/FindTBB.cmake:190 (file):
  file failed to open for reading (No such file or directory):

    C:/Users/lf/Dev/slic3r-devstuff/destdir-64-dev/usr/local/include/tbb/tbb_stddef.h
Call Stack (most recent call first):
  CMakeLists.txt:246 (find_package)

-- Found TBB: C:/Users/lf/Dev/slic3r-devstuff/destdir-64-dev/usr/local/include (found version ".")
--   TBB_FOUND               = TRUE
--   TBB_INCLUDE_DIRS        = C:/Users/lf/Dev/slic3r-devstuff/destdir-64-dev/usr/local/include
--   TBB_DEFINITIONS         =
--   TBB_LIBRARIES           = C:/Users/lf/Dev/slic3r-devstuff/destdir-64-dev/usr/local/lib/tbb_static.lib
--   TBB_DEFINITIONS_DEBUG   = -DTBB_USE_DEBUG=1
--   TBB_LIBRARIES_DEBUG     = C:/Users/lf/Dev/slic3r-devstuff/destdir-64-dev/usr/local/lib/tbb_static_debug.lib
--   TBB_DEFINITIONS_RELEASE =
--   TBB_LIBRARIES_RELEASE   = C:/Users/lf/Dev/slic3r-devstuff/destdir-64-dev/usr/local/lib/tbb_static.lib
-- Found CURL: optimized;C:/Users/lf/Dev/slic3r-devstuff/destdir-64-dev/usr/local/lib/libcurl_a.lib;debug;C:/Users/lf/Dev/slic3r-devstuff/destdir-64-dev/usr/local/lib/libcurl_a_debug.lib
-- qhull Version: 7.2.0 (static linking)
-- Parallelization with Intel TBB
CMake Error at C:/Program Files/CMake/share/cmake-3.14/Modules/FindBoost.cmake:1373 (file):
  file STRINGS file
  "C:/Users/lf/Dev/slic3r-devstuff/destdir-64-dev/usr/local/include/boost-1_63/boost/version.hpp"
  cannot be read.
Call Stack (most recent call first):
  src/libnest2d/include/libnest2d/backends/clipper/CMakeLists.txt:60 (find_package)

CMake Error at C:/Program Files/CMake/share/cmake-3.14/Modules/FindBoost.cmake:2132 (message):
  Unable to find the requested Boost libraries.

  Boost version: 0.0.0

  Boost include path:
  C:/Users/lf/Dev/slic3r-devstuff/destdir-64-dev/usr/local/include/boost-1_63

  Detected version of Boost is too old.  Requested version was 1.58 (or
  newer).
Call Stack (most recent call first):
  src/libnest2d/include/libnest2d/backends/clipper/CMakeLists.txt:60 (find_package)

CMake Warning (dev) at src/libnest2d/include/libnest2d/backends/clipper/CMakeLists.txt:73 (target_link_libraries):
  Policy CMP0079 is not set: target_link_libraries allows use with targets in
  other directories.  Run "cmake --help-policy CMP0079" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  Target

    libnest2d

  is not created in this directory.  For compatibility with older versions of
  CMake, link library

    ClipperBackend

  will be looked up in the directory in which the target was created rather
  than in this calling directory.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found NLopt in 'C:/Users/lf/Dev/slic3r-devstuff/destdir-64-dev/usr/local/lib'.
-- Using NLopt include directory 'C:/Users/lf/Dev/slic3r-devstuff/destdir-64-dev/usr/local/include'.
-- Using NLopt library 'C:/Users/lf/Dev/slic3r-devstuff/destdir-64-dev/usr/local/lib/nlopt.lib'.
CMake Warning (dev) at src/libnest2d/include/libnest2d/optimizers/nlopt/CMakeLists.txt:61 (target_link_libraries):
  Policy CMP0079 is not set: target_link_libraries allows use with targets in
  other directories.  Run "cmake --help-policy CMP0079" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  Target

    libnest2d

  is not created in this directory.  For compatibility with older versions of
  CMake, link library

    NloptOptimizer

  will be looked up in the directory in which the target was created rather
  than in this calling directory.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- WXWIN environment set to:
CMake Error at C:/Program Files/CMake/share/cmake-3.14/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
  wxWidgets_INCLUDE_DIRS base core adv html gl) (Required is at least version
  "3.1")
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.14/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.14/Modules/FindwxWidgets.cmake:990 (find_package_handle_standard_args)
  src/CMakeLists.txt:51 (find_package)

-- Configuring incomplete, errors occurred!
See also "C:/Users/lf/Dev/slic3r/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/lf/Dev/slic3r/build/CMakeFiles/CMakeError.log".
```

lf-, 2019-04-06:
Note: this error can be caused by forgetting `Win64` in the generator option for cmake. Now for the next error. I just need to point Boost to the right spot and maybe it will work.

closed 2019-04-06 by lf-

lf-, 2019-04-06:
Further investigation reveals that the static libraries provided for old VS2013 aren't usable by the newer versions. Argh. So now I have to compile all that stuff.

Please ship newer ones.

VictorBI3D, 2019-10-01:
I have this same problem with "Visual Studio 16 2019", i try solve the problem but without success, do you have idea what can be?

`
WXWIN environment set to: 
CMake Error at C:/Program Files/CMake/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find wxWidgets (missing: wxWidgets_LIBRARIES
  wxWidgets_INCLUDE_DIRS html adv gl core base) (Required is at least version
  "3.1")
Call Stack (most recent call first):
  C:/Program Files/CMake/share/cmake-3.15/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  C:/Program Files/CMake/share/cmake-3.15/Modules/FindwxWidgets.cmake:990 (find_package_handle_standard_args)
  src/CMakeLists.txt:51 (find_package)`