prusa3d/PrusaSlicer #2080: [Feature Request] automatic sd card handling

quillford, 2019-04-06:
It would be nice if slic3r automatic handled sd cards like daid's original cura did. Basically, when an SD card was detected, a dedicated button would appear to send the gcode to the sd card. After it saved on the sd, it would have a dialog allowing you to eject the sd card. This last part could just be a setting to enable/disable automatic ejecting. 

I may implement this myself. If I did, would a PR for it be merged?

bubnikv, 2019-04-08:
>  I may implement this myself. If I did, would a PR for it be merged?

It depends on the code quality and the amount of changes. Minimal pull
request and / or pull request, where the integration into the Slic3r has a
clearly defined minimal boundary (little change in the existing code) have
a higher chance to be merged.

We are certainly interested in this feature, just we are too busy right now
to review and test extensive pull requests. After the current release we
will have more time for that.

dantman, 2019-05-26:
I'm also very interested in this feature. The current workflow on OSX is getting rather tedious. Insert SD, Export G-Code, Select folder, open finder (often resulting in OSX focusing a finder window on a different desktop space), eject SD, return to or close PrusaControl.

The need to leave the slicer (and possibly the desktop space) just to eject the SD card is the most annoyinng part.

rtyr, 2020-04-17:
Implemented. Closing.

closed 2020-04-17 by rtyr