prusa3d/PrusaSlicer #2081: Place on face makes part selection weird

Sidic101, 2019-04-06:
### Version
1.42 beta 1

### Operating system type + version
Windows 10 pro 64bit Version	10.0.17763 Build 17763

### Behavior
* _Describe the problem_
when the "place on face" interface is running, you sometimes need to select objects twice.
* _Steps needed to reproduce the problem_
Add multiple models. Click on one of them. Click the "place on face" button. Click another object. This results in the second model being selected for an extremely short time (a few ms) and than deselected and you also need to click the "place on face" button again to use it for the second part.
*_Is this a new feature request?_
No, because in alpha 7 this worked normally (the part would stay selected and Place on bed would stay on.

enricoturri1966, 2019-04-08:
This issue may have been fixed by https://github.com/prusa3d/Slic3r/commit/3cd6db2278a92eab6067c13f81b9da90b8540024

FidelCapo, 2019-04-08:
Thank you for report. This issue should be fixed in next release.
Closing.

closed 2019-04-08 by FidelCapo