prusa3d/PrusaSlicer #2082: Add M73 P100 when finished

guestisp, 2019-04-06:
Currently, when print is finished, the progress bar on lcd is still showing 99%. This because in gcode there isn't any `M73 P100`. The last progress update is `M73 P99`

Yes, we can add the P100 to the end g-code, but I think this is a sort of bug or something to improve. It doesn't make sense to update a progress bar only up to 99%.....

enricoturri1966, 2019-04-08:
Working on it -> http://dev.prusa3d.com/browse/SPE-897

guestisp, 2019-04-08:
> Working on it -> http://dev.prusa3d.com/browse/SPE-897

Can't open the issue on Jira. Isn't it public ?

bubnikv, 2019-04-08:
that's for us for reference.

FidelCapo, 2019-04-09:
Thank you for report. This issue should be fixed in next release.
Closing.

closed 2019-04-09 by FidelCapo

guestisp, 2019-04-09:
Great!