prusa3d/PrusaSlicer #2086: Changing settings running on a networked filesystem is extremely slow

zbrozek, 2019-04-07:
### Version
Slic3rPE-1.42.0-beta1

### Operating system type + version
Windows 10 build 17134

### Behavior
My home has multiple users and multiple computers, so in order to keep everyone using the same slicing settings and software I run slic3r from an SMB share. This generally works great over the wired network. It works OK from the wireless network. It's absolutely awful when over the broader internet.

As far as I can tell, slic3r isn't fully resident in memory or isn't caching config files. There are very long pauses both when clicking into a setting box, then again when changing it. It would be super wonderful if slic3r were snappy no matter where it's stored, perhaps reading/writing to disk only when truly necessary (e.g., on open, saving settings, or close).

bubnikv, 2019-04-08:
The only thing that being constantly written to is slic3r.ini to save the
names of the last profiles selected. Everything else is being cached into
RAM, nothing is re-read.
We may make the automatic saving of slic3r.ini configurable.

bubnikv, 2019-04-08:
You may try some of the tools by
https://docs.microsoft.com/en-us/sysinternals/downloads/sysinternals-suite
(I think filemon will allow you to watch for file access online)

bubnikv, 2019-04-13:
Please help us debug your network usage. Without your help, we are going to close this issue.

zbrozek, 2019-04-13:
Sorry; missed your reply earlier. I used "Process Monitor" and filtered on the process "slic3r.exe". The output is fascinating (at least to someone unused to examining the behavior of large applications). Some observations:

- slic3r startup is surprisingly slow (many seconds) even when executed locally, but remote startup is extremely slow (tens of seconds to a minute over a reasonably fast internet connection).
- Some of the slowness seems to be due to Windows client-side caching. I can't tell if it's by design or misbehavior, but it tries (and fails with NAME NOT FOUND) repeatedly to create `C:\Windows\CSC\v2.0.6\namespace\fileserver` at many points during startup, consuming 20-50 milliseconds each time.[1]
- There are many consecutive events that are reading slic3r.dll at various offsets. It looks like that is deliberately non-cached[2] and causing a lot of back-and-forth file-transfer latency.
- There are huge numbers (many hundreds) of writes to slic3r.ini that are 2-4 bytes long and consume about 20 milliseconds (which is about my round-trip internet latency right now) each. Each block of these writes is bookended by the above-mentioned caching error, which itself adds another 20-50 milliseconds to each block.
- Upon editing any setting, there are multiple network hits to preview.svg, but it doesn't seem to account for all of the perceived latency (seconds per click or keystroke) when interacting with the UI.
- It's worth noting that I'm filtering on slic3r.exe. But there's a crazy amount of event traffic every time I go to edit a field. By volume a lot of it is from ctfmon.exe querying large numbers of registry keys under `HKLM\SOFTWARE\Microsoft\Input\Locales` repetitively as it iterates through `HKCU\Software\Microsoft\Input\Locales\loc_*`. I don't see why that would be different over a network vs locally, but maybe that traffic is burying whatever is actually taking time.

[1] The detail field for this says `Desired Access: Read EA, Write EA, Read Attributes, Write Attributes, Delete, Read Control, Synchronize, Disposition: Open, Options: Synchronous IO Non-Alert, Attributes: n/a, ShareMode: Read, Write, Delete, AllocationSize: n/a`
[2] Flags are typically like `Offset: 6,288,384, Length: 4,096, I/O Flags: Non-cached, Paging I/O, Synchronous Paging I/O, Priority: Normal` but that may not actually mean what I am assuming them to mean. I'm no expert.

bubnikv, 2019-04-13:
Thanks for the detailed log.

oh, I thought you are just pulling the slic3r profiles from the network, not the whole Slic3r installation.

we should review the slic3r.ini access

bubnikv, 2019-04-13:
> we should review the slic3r.ini access

but I am afraid it will not help much at your setup

zbrozek, 2019-04-13:
Indeed; the entire installation (including profiles) is network-side so that I only have to maintain one location rather than a half dozen. It does seem to accentuate performance bottlenecks.

Improving start-up time would certainly be welcome, and it seems likely that larger block reads (e.g., loading slic3r.ini entirely in a single sequential read, then caching it) would be a pretty big improvement for everybody. Similarly, storing all of the resources in a single archive and loading the whole thing at a stroke would be another substantial start-up win. The per-file-open latency seems to overwhelm the data copy time.

Do you know why entering any text field loads many-hundreds of locale-related registry keys?

bubnikv, 2019-04-13:
>  Do you know why entering any text field loads many-hundreds of
locale-related registry keys?

No. I have no idea, whether it is Windows or wxWindows issue.

We are not re-loading slic3r.ini, but we are writing it every time
something changes (for example the last selected profile). If we are doing
it more often, then we should fix it.

vojtechkral, 2019-04-15:
Looking into `strace` output, I'm not seeing anything terribly out of ordinary regarding `slic3r.ini`. There does seem to be a problem with loading some of the png icons, which seem to be loaded repeatedly waaay too many times. We should probably fix that. Other than that, nothing too horrible, although of course there's always room for improvement.

Regarding running slic3r from a networked drive isn't really a scenario we optimize for very much, _especially_ over broad internet. I believe an average application can't be expected to work well over a slow connection. It would probably be best to synchronize the intallation in some other, less realtime-y way.

seanmcne, 2023-01-05:
It looks like this could be linked to or solved by   resolution to this issue  https://github.com/prusa3d/PrusaSlicer/issues/1110
 
Also, I think the ability to run a script or load a profile/settings from an alternative/shared location would be great - we have several PCs in our home and we end up having quite the variety of settings which would be better to “keep in sync” for more consistent printing.

closed 2026-09-29 by prusaslicer-bot[bot]