prusa3d/PrusaSlicer #2088: wipe at print end into air --> missing FAQ / -link

akust, 2019-04-07:
I also have the problem of "wipes into thin air", and found your short answers ("nobug, close, faq"), well.. as Palette2 user "close the issue" doesn't help me, so I searched for FAQ but couldn't find it..

would you be so kind and add a link to the FAQ? 
thank you!

bubnikv, 2019-04-08:
It is the final ramming to get the excess of the molten filament out of the melt zone before pulling the filament out. You are right we should document it somewhere. Ideally, our content guys would whip up a Slic3r manual.

Closing.

bubnikv, 2019-04-08:
References https://github.com/prusa3d/Slic3r/issues/1355#issuecomment-480632799

bubnikv, 2019-04-13:
Added to https://github.com/prusa3d/Slic3r/wiki/Frequently-Asked-Qeustions-(FAQ)
Closing

closed 2019-04-13 by bubnikv